node-opcua/node-opcua #40: MonitoredItem on changed not fired

MackyNacky, 2015-01-20:
In UAExpert i see the attributes are changed realtime. But using this module the on changed event is not fired. Also console.log("keepalive"); is not printed.  Somebody have any suggestions? Thank you in advance...

Code that works (getting variable value 1 time):

    function(callback) {
       the_session.readVariableValue("ns=2;s=WINDSAMP/Variables/fWindspeed1_1s_var", function(err,dataValues) {
           if (!err) {
               console.log(" Winddirection 1 = " , dataValues);
           }
           callback(err);
       });
    }
Code tha does not work:

    function(callback) {
       the_subscription=new opcua.ClientSubscription(the_session,{
           requestedPublishingInterval: 500,
           requestedLifetimeCount: 1000,
           requestedMaxKeepAliveCount: 10,
           maxNotificationsPerPublish: 10,
           publishingEnabled: true,
           priority: 0
       });
       
       the_subscription.on("started",function(){
           console.log("subscription started for 2 seconds - subscriptionId=",the_subscription.subscriptionId);
       }).on("keepalive",function(){
           console.log("keepalive");
       }).on("terminated",function(){
           callback();
       });
       
       setTimeout(function(){
           the_subscription.terminate();
       },10000);
       
		var monitoredItem  = the_subscription.monitor({
		      nodeId: opcua.resolveNodeId("ns=2;s=WINDSAMP/Variables/fWindspeed1_1s_var"),
		      attributeId: 13
		  },
		  {
		      samplingInterval: 100,
		      discardOldest: true,
		      queueSize: 10
		  },
		  opcua.read_service.TimestampsToReturn.Both
		);   
		
		console.log("-------------------------------------");
 
       monitoredItem.on("changed",function(dataValue){
          console.log(" Windspeed = ",dataValue.value.value);
       });

    },

erossignon, 2015-01-20:
Hi MackyNacky, 
can you be more specific,What is the server your are trying to connect to ?
This will help to reproduce the behavior

MackyNacky, 2015-01-21:
Hi Erossignon,

This is the full code:
Hi Etienne,

Thank you for you fast reaction on github. Sorry that i contact you on this way. But connecting to the server needs username and password. I cannot publish that on github:(.

This is my full code:

/*global require,console,setTimeout */
var opcua = require("node-opcua");
var async = require("async");
var s = require("./node_modules/node-opcua/lib/datamodel/structures");

var client = new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://" + "188.200.226.106:4840";

var the_session, the_subscription;

var userIdentityToken = new s.UserNameIdentityToken({
    policyId: "",
    userName: "",
    password: "",
    encryptionAlgorithm: ""
});

async.series([

    // step 1 : connect to
    function(callback)  {
        client.connect(endpointUrl,function (err) {
            if(err) {
                console.log(" cannot connect to endpoint :" , endpointUrl );
            } else {
                console.log("connected !");
            }
            callback(err);
        });
    },

    // step 2 : createSession
    function(callback) {
        client.createSession(userIdentityToken, function(err,session) {
            if(!err) {
                the_session = session;
                console.log(session);
            }
            callback(err);
        });
    },

    // step 3 : browse
/*
    function(callback) {
       the_session.browse("RootFolder", function(err,browse_result){
           if(!err) {
               browse_result[0].references.forEach(function(reference) {
                   console.log( reference.browseName);
               });
           }
           callback(err);
       });
    },
*/

    // step 4 : read a variable with readVariableValue
    function(callback) {
       the_session.readVariableValue("ns=2;s=WINDSAMP/Variables/fWindspeed1_1s_var", function(err,dataValues) {
           if (!err) {
               console.log(" Winddirection 1 = " , dataValues);
           }
           callback(err);
       });
    },
    
    // step 4' : read a variable with read
    function(callback) {
       var max_age = 0;
       var nodes_to_read = [
          { nodeId: "ns=2;s=WINDSAMP/Variables/fWindspeed1_1s_var", attributeId: 13} 
       ];
       the_session.read(nodes_to_read, max_age, function(err,nodes_to_read,dataValues) {
           if (!err) {
               console.log(" Winddirection 2 = " , dataValues[0]);
           }
           callback(err);
       });
    },
    
    // step 5: install a subscription and install a monitored item for 10 seconds
    function(callback) {
       the_subscription=new opcua.ClientSubscription(the_session,{
           requestedPublishingInterval: 500,
           requestedLifetimeCount: 1000,
           requestedMaxKeepAliveCount: 10,
           maxNotificationsPerPublish: 10,
           publishingEnabled: true,
           priority: 1
       });
       
       the_subscription.on("started",function(){
           console.log("subscription started for 2 seconds - subscriptionId=",the_subscription.subscriptionId);
       }).on("keepalive",function(){
           console.log("keepalive");
       }).on("terminated",function(){
           callback();
        });
       
       setTimeout(function(){
           the_subscription.terminate();
       },10000);
       
       
       console.log("nodeId: " + opcua.resolveNodeId("ns=2;s=WINDSAMP/Variables/fWindspeed1_1s_var"));
       
		var monitoredItem  = the_subscription.monitor({
		      nodeId: opcua.resolveNodeId("ns=2;s=WINDSAMP/Variables/fWindspeed1_1s_var"),
		      attributeId:13
		  },
		  {
		      samplingInterval: 100,
		      discardOldest: true,
		      queueSize: 10
		  },
		  opcua.read_service.TimestampsToReturn.Both
		);   
		
		console.log("-------------------------------------");
       
       monitoredItem.on("changed",function(dataValue){
          console.log(" Windspeed = ",dataValue.value.value);
       });

    },

    // close session
    function(callback) {
        the_session.close(function(err){
            if(err) {
                console.log("session closed failed ?");
            }
            callback();
        });
    }

],
function(err) {
    if (err) {
        console.log(" failure ",err);
    } else {
        console.log("done!");
    }
    client.disconnect(function(){});
}) ;

erossignon, 2015-01-21:
After investigation I found that the server send a ServiceFault message with Bad_Timeout Error in place of each expected Publish responses.

as started in the spec (Spec 1.02 part 4 page 81 5.13.2.2 Function DequeuePublishReq()) 
<blockquote>the server will dequeue the PublishRequest  in first-in first-out order and will validate if the publish request is still valid by checking the timeoutHint in the RequestHeader. If the request timed out, the server will send a Bad_Timeout service result for the request and de-queue another publish request.
</blockquote>

In Part 4. page 144 
Request Header the timeoutHint is described this way
 timeoutHint UInt32 
<blockquote>This timeout in milliseconds is used in the Client side Communication Stack to set the timeout on a per-call base. For a Server this timeout is only a hint and can be used to cancel long running operations to free resources. If the Server detects a timeout, he can cancel the operation by sending the Service result Bad_Timeout. The Server should wait  at minimum the timeout after he received the request before cancelling the operation. <strong>The value of 0 indicates no timeout.
</strong></blockquote>

Node-opcua uses timeoutHint=0 in the requestHeader by default.
In our case , the server doesn't consider a timeout=0 means no timeout as per the spec.
This is clearly a bug on the server side, that you might want to report to the vendor.

Anyway, I have put a workaround in the code, that should fix the issue.
please confirm.

erossignon, 2015-01-29:
MackyNacky said : Thank you very much it works :100: % :)
Closing

closed 2015-01-29 by erossignon