node-opcua/node-opcua #43: NodeCrawler - crawl and fill tree-like object with nodes hierachy

Jochen1980, 2015-01-26:
Hi, 

currently I am playing around with node crawler. I would like to browse through all nodes and fill an object with the hierachy of all browsed nodes. Summing up, I try to build the tree of i.e. UAExpert in nodejs-opcua. Any ideas or code snippets how to solve this issue?

erossignon, 2015-01-28:
There is a example in the simple_client:
you need to activate the crawler by passing the ```--crawl``` option to the command line:

```
$ node bin\simple_client --endpoint opc.tcp://.... --crawl
```

the code snippet is the following : 

```javascript
            var crawler = new NodeCrawler(the_session);

            crawler.on("browsed",function(element){
                console.log("->",element.browseName.name,element.nodeId.toString());
            });

            var nodeId = "ObjectsFolder";
            console.log("now crawling object folder ...please wait...");
            crawler.read(nodeId, function (err, obj) {
                if (!err) {
                    treeify.asLines(obj, true, true, function (line) {
                        console.log(line);
                    });
                }
                callback(err);
            });

```

closed 2016-03-12 by erossignon

andreafresa, 2020-03-17:
Hello,
I tried to run $ node bin\simple_client --endpoint opc.tcp://myserver --crawl but  I get this error.
ERROR !!! callback has thrown en error  ReferenceError: ApplicationType is not defined
 Any idea?