node-opcua/node-opcua #45: Creating methods in objects

limjunliang, 2015-01-29:
Hi, in the released version, the add method to the objects is lacking. However, in the codes that are in development, I notice addMethod() function has been implemented in the server-engine. I tried implementing it but to no avail.

Currently, I have this error: 
Cannot find module '_generated_/_auto_generated_CallMethodRequest'

How do I generate this file?

erossignon, 2015-01-29:
Hi limjunliang, 
you need to run ````npm install``` after ```git pull``` to make sure to regenerate required files
Let me know if this fixes your issue.

limjunliang, 2015-01-30:
Thanks! It's working.

limjunliang, 2015-01-30:
when will the next release be with the methods feature?

limjunliang, 2015-01-30:
I am new to javascript and nodejs so it'll be great to have assistance on this issue.
I tested with the add method but I am having difficulty with the calling of methods using UAExpert.
No input and output arguments are displayed. The arguments within the methods are shown in blue in Prosys. But in my tested server, they are shown as variables in green. What does the blue icon refer to?

[Desired result: Retrieved from Prosys sample server](https://cloud.githubusercontent.com/assets/2024408/5973747/ea8dc1f4-a8a6-11e4-935f-0edc0d32daa4.png)

[Problem: Nothing is displayed in Input and Output arguments]
(https://cloud.githubusercontent.com/assets/2024408/5973753/06414394-a8a7-11e4-9c59-305e6d94a285.png)

Attached below is my server code:

```javascript
var opcua = require("node-opcua");

var datatype = opcua.DataType;
var statusCode = opcua.StatusCodes;
var variantArrayType = require("node-opcua/lib/datamodel/variant").VariantArrayType;
var CallMethodResult = require("node-opcua/lib/services/call_service").CallMethodResult;
var method;

// Let's create an instance of OPCUAServer
var server = new opcua.OPCUAServer({
    port: 6001 // the port of the listening socket of the server
});

// we can set the buildInfo
server.buildInfo.productName = "MySampleServer1";
server.buildInfo.buildNumber = "7658";
server.buildInfo.buildDate = new Date(2014, 5, 2);

function post_initialize() {
    console.log("initialized");
    function construct_my_address_space(server) {

        // create folder structure
        server.engine.createFolder("RootFolder", {browseName: "Device"});
        server.engine.createFolder("Device", {browseName: "Diagnostics"});
        server.engine.createFolder("Diagnostics", {browseName: "Message history"});

        //create object in server
        var object = server.engine.addObjectInFolder(server.engine.findObject("ObjectsFolder"), {
            browseName: "myObject",
            nodeId: "ns=1;s=MyObject"
        });

        method = server.engine.addMethod(object, {
            browseName: "getMessage",
            inputArguments: [
                {
                    name: "messageId",
                    description: {text: "pass a message id"},
                    dataType: datatype.String
                }
            ],
            outputArguments: [
                {
                    name: "result",
                    description: {text: "message retrieved based on the passed in id"},
                    dataType: datatype.String,
                    valueRank: 1
                }
            ]
        });

        server.engine.bindMethod(method.nodeId, getMessage.bind(server.engine));
    }

    function getMessage(inputArguments, context, callback) {
        var nbBarks = inputArguments[0].value;
        console.log("Hello World ! I will bark ", nbBarks, "times");
        var barks = [];
        for (var i = 0; i < nbBarks; i++) {
            barks.push("Whaff");
        }
        var callMethodResult = new CallMethodResult({
            statusCode: statusCode.Good,
            outputArguments: [
                {
                    dataType: datatype.String,
                    arrayType: variantArrayType.Array,
                    value: barks
                }
            ]
        });
        callback(null, callMethodResult);
    }

    construct_my_address_space(server);

    server.start(function () {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        console.log("port ", server.endpoints[0].port);
        //var endpointUrl = server.endpoints[0].endpointDescription().endpointUrl;
        //console.log(" the primary server endpoint url is ", endpointUrl );

        var object = server.engine.findObjectByBrowseName("myObject");
        object.getMethodById(method.nodeId);

       // now call it
        var inputArguments = [{dataType: datatype.UInt32, value: 3}];
        var context = {};

        method.execute(inputArguments, context, function (err, callMethodResponse) {
            console.log(callMethodResponse.statusCode);
            console.log(callMethodResponse.outputArguments.length);
            console.log(" Result = ", callMethodResponse.outputArguments[0].value);
        });
    });
}
server.initialize(post_initialize);
```

erossignon, 2015-02-01:
Hi limjunliang, Thank you for your valuable feedback that helped me to nail down a bug.
Please let me know if this is now sorted with the most recent version on the repository

limjunliang, 2015-02-02:
You are welcome. I tried but now when I call the method to open the dialog through the UAExpert, the whole UAExpert application stops responding. The console output is attached as follows. May I check with you the entry point in the code when UAExpert calls the method?

[UAExpert client stops responding.]()

[Console output]()

erossignon, 2015-02-02:
Hi limjunliang, I do not have the same behavior on my end with UAExperts
[image]

I would suggest that you try to run ```npm install``` again so that the _generated_ files affected by the latest commits are regenerated.
It's a good idea to run ```npm install```   after fetching a update from the git repository.
Let me know ..

limjunliang, 2015-02-02:
Hi, I did do a npm install before posting the issue. I will try it again. May I also check with you the version of UAExpert you are using? Thanks.

limjunliang, 2015-02-02:
I just did a checkout of the project from git repository and a npm install. Same issue with the UAExpert client not responding. However, when I tried with the version of the code a few days ago, the method call dialog could be open.

erossignon, 2015-02-02:
I am on UAExpert version 1.2.2.175 , and just find out that V1.3 upgrade is now available, 
Let me install it and come back to you.

limjunliang, 2015-02-02:
I tried with UAExpert 1.2 and it is working. It seems like it has problems with V1.3.

erossignon, 2015-02-02:
limjunliang, I confirm that  UAExpert  1.3.0.201 is crashing , whereas  UAExpert version 1.2.2.175 seems to work as expected.
Investigating ...

erossignon, 2015-02-02:
limjunliang: I found a fix, you have to add ```valueRank: -1,``` in the InputArgument definition, like this:

```javascript
        method = server.engine.addMethod(object, {
            browseName: "getMessage",
            inputArguments: [
                {
                    name: "messageId",
                    description: {text: "pass a message id"},
                    valueRank: -1,    //<<<<< ADD THIS LINE
                    dataType: datatype.String
                }
            ],
            outputArguments: [
                {
                    name: "result",
                    description: {text: "message retrieved based on the passed in id"},
                    dataType: datatype.String,
                    valueRank: 1
                }
            ]
        });
```

erossignon, 2015-02-02:
for info here is what the specs says about valueRank:
```javascript
    /**
     * valueRank (5.6.2 Variable NodeClass part 3)
     * This Attribute indicates whether the Value Attribute of the Variable is
     * an array and how many dimensions the array has.
     * It may have the following values:
     *    n > 1: the Value is an array with the specified number of dimensions.
     *    OneDimension        (1): The value is an array with one dimension.
     *    OneOrMoreDimensions (0): The value is an array with one or more dimensions.
     *    Scalar             (−1): The value is not an array.
     *    Any                (−2): The value can be a scalar or an array with any number of
     *                             dimensions.
     *   ScalarOrOneDimension(−3): The value can be a scalar or a one dimensional array.
     *
     *   NOTE: All DataTypes are considered to be scalar, even if they hav
     *   array-like semantics like ByteString and String.
     */
```

closed 2015-02-04 by erossignon

SeetaiahB, 2017-07-18:
Hi, Can you give an example , how to set input argument and output arguments for a call method.
I have 
[image]

[image]

[image]

I am familar with access data value from simple node but i never used calling a method.

Appreciated posting an example how to call a method constructing input and out arguments.

thanks