node-opcua/node-opcua #47: Node OPC Client constructor not found for expandedId

lvchaoin, 2015-02-01:
While running the subscription, it sometimes throw out the following error, and it will cause the subscription suspended, would you please suggest what happend ?
I'm using the opc foundation's Com Opc Server Wrapper with Kepeware 4 to monitor a simulator tag configured in kepware 4. If I tried to reboot the nodejs process for several times, then the error will disappear, but I don't know how to make it more stable.
Thanks a lot !
```
cannot find constructor for expandedId
{ identifierType: { key: 'NUMERIC', value: 1 },
  value: 820,
  namespace: 0 }
```

erossignon, 2015-02-01:
This NodeId matches the Encoding DefaultBinary of the StatusChangeNotification which is currently not supported.. We shall address this.

lvchaoin, 2015-02-07:
Hi erossignon,
I'm currently developing a web client using the nodeopcua, and I have been troubled in this problem for several days, I spent much time to debug and tried to solve the problem, but failed, I guess there should be some extra js files need to be added by the codegen to support the requirement that I want. I was confused when debugged into the constructorMap(it's a object, but I don't know how to view the values in this object, only can see some expression such as '449 in constructorMap' will return true, some others will return false, the js file I debugged is factories_factories.js)
could you kindly suggest is there any trick I can do to ignore the error and can make me keep on going ? Or could you please suggest what should be the general steps to solve this issue ? Cause I have learned the opc and node only for a few weeks, your suggestion should be very helpful for me even if I want to solve the problem by myself.

Sorry to disturb.
Best Regards,

lvchaoin, 2015-02-09:
thanks for the fix, I tried and it works well.

closed 2015-02-09 by lvchaoin