node-opcua/node-opcua #53: Null Pointer Exception in CreateSession() - server not defined

Jochen1980, 2015-02-05:
Hi, I tried to create a simple client. So I started to write some kind of main-Method. After I did the connect, I want to create a session. Unfortunately I got an error: 
    ... \node-opcua\lib\client\opcua_client.js:638
    this.serverUri = endpoint.server.applicationUri;
TypeError: Cannot read property 'server' of undefined at OPCUAClient._createSession (... \node_modules\node-opcua\lib\client\opcua_client.js:639:30)
    at OPCUAClient.createSession

Here is the corresponding code snippet:
```
console.log("BP0");
var client		= new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://" + require("os").hostname() + ":48010";
client.connect(endpointUrl, function(err){
	if (err) {
		console.log(" failure ",err);
	} else {
		console.log("done!")
	}
});
console.log("Connected!");
var userIdentityToken = new s.AnonymousIdentityToken({policyId: "0"});
client.createSession(userIdentityToken, function(err, ses){
    if(!err){
        session = ses;
    } else {
    	console.log("ERROR 100");
    }
});
console.log("Session created!");
client.disconnect(function(err){
	if (!err){
		console.log("Disconnected!")
	} else {
		console.log("ERROR 101");
	}
});
console.log("Disconnected!");
console.log("BP1");
```

erossignon, 2015-02-06:
Hi Jochen,
remember that nodejs is a asynchronous framework.
the create session can only start when the client is connected  and so on,
so you need to trigger the action inside your callback function not outsite.
therefore the correct sequence is

```javascript
console.log("BP0");
var client      = new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://" + require("os").hostname() + ":48010";

client.connect(endpointUrl, function(err) {
    if (err) {
        console.log(" failure ",err);
    } else {
        console.log("Connected!");
        var userIdentityToken = new s.AnonymousIdentityToken({policyId: "0"});
        client.createSession(userIdentityToken, function(err, ses){
            if(!err){
               session = ses;
               console.log("Session created!");
            } else {
               console.log("ERROR 100");
           }
           client.disconnect(function(err){
           if (!err){
              console.log("Disconnected!")
          } else {
                console.log("ERROR 101");
          }
         console.log("BP1");
     });
});
```
This is the theory, but as you can see the code can become easily messy , so you'll need to use third party librairies such as "async" to make it more friendly, (see ```simple_client```)

Jochen1980, 2015-02-06:
Thanks a lot Etienne!

Jochen1980, 2015-02-11:
Hi Etienne,

out of curiosity I wanted to finish my main-function trial 'the messy way'. I hit 'BP5' so I created a session, but the session seems not valid, see the log message received :  
```
{ [Error:  ServiceFault returned by server {
 "responseHeader": {
  "timeStamp": "2015-02-11T10:23:21.119Z",
  "requestHandle": 7,
  "serviceResult": {
   "value": 2149908480,
   "description": "The session id is not valid.",
   "name": "BadSessionIdInvalid"
  },
  "serviceDiagnostics": {},
  "stringTable": []
 }
}]
```
The corresponding code: 
```javascript
console.log("BP0");
var client		= new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://" + require("os").hostname() + ":48010";
var session = null;
client.connect(endpointUrl, function(err){
	if (err) {
		console.log("Error : trying to connect ... ",err);
	} else { // connected
		console.log("Connected!");
		var userIdentityToken = new s.AnonymousIdentityToken({policyId: "0"});
		client.createSession(userIdentityToken, function(err, ses){
		    if(err){
		    	console.log("Error : trying to create session  ... ", err);
		    	client.disconnect(function(err){ // disconnect
					if (err){ console.log("Error : trying to disconnect ... ",err); 
					} else { console.log(" disconnected! "); }
				});
		    } else { // session available
		    	console.log("Session created!");
		    	session = ses;
		    	session.browse("ObjectsFolder", function(err, res, diagnostics){
					if(err){
						console.log("Error : trying to browse  ... ", err);
						session.close(function(err){ // close session
				            console.log(" session closed");
				        });
						client.disconnect(function(err){ // disconnect
							if (err){ console.log("Error : trying to disconnect ... ",err); 
							} else { console.log(" disconnected! "); }
						});
					} else {
						console.log("BP5");
						JSON.stringify(res, null, 4);
						session.close(function(err){ // close session
				            console.log(" session closed");
				        });
						client.disconnect(function(err){ // disconnect
							if (err){ console.log("Error : trying to disconnect ... ",err); 
							} else { console.log(" disconnected! "); }
						});
					}
		    	});
		    }
		});
		console.log("BP9");
	}
});
console.log("BP1");
```

erossignon, 2015-02-11:
Hi Jochen
````
                        session.close(function(err){ // close session
                            console.log(" session closed");
                        });
                        client.disconnect(function(err){ // disconnect
                            if (err){ console.log("Error : trying to disconnect ... ",err); 
                            } else { console.log(" disconnected! "); }
                        });
```

should probably be
````
                        session.close(function(err){ // close session
                            console.log(" session closed");
                            client.disconnect(function(err){ // disconnect
                                if (err){ console.log("Error : trying to disconnect ... ",err); 
                                } else { console.log(" disconnected! "); }
                            });
                        });
```

Jochen1980, 2015-02-16:
Thanks Etienne, this was the solution!

closed 2015-02-16 by Jochen1980