node-opcua/node-opcua #55: Browserify node-opcua

zbysiusiekiera, 2015-02-13:
Hello,

I am trying to browserify the node-opcua module withouth success. Is it possible in the current stage of developement?  I have tried setting transform option with requirish module but still it can't resolve the modules. File sample_client.js contains one line wich basically just requires the node-opcua module.

It works if I run it as 'node sample_client.js'. Browserify command works with other modules (like async). The error I am receiving:

<pre>
$ browserify -t requirish ./opcua/sample_client.js > bundle.js 
Error: Cannot find module 'lib/datamodel/structures' from '/home/jkochman/Work/MTViewCenter2/node_modules/node-opcua'
    at /usr/local/lib/node_modules/browserify/node_modules/browser-resolve/node_modules/resolve/lib/async.js:51:17
    at process (/usr/local/lib/node_modules/browserify/node_modules/browser-resolve/node_modules/resolve/lib/async.js:159:43)
    at /usr/local/lib/node_modules/browserify/node_modules/browser-resolve/node_modules/resolve/lib/async.js:168:21
    at load (/usr/local/lib/node_modules/browserify/node_modules/browser-resolve/node_modules/resolve/lib/async.js:99:43)
    at /usr/local/lib/node_modules/browserify/node_modules/browser-resolve/node_modules/resolve/lib/async.js:105:22
    at /usr/local/lib/node_modules/browserify/node_modules/browser-resolve/node_modules/resolve/lib/async.js:22:47
    at Object.oncomplete (fs.js:107:15)
</pre>

Kind regards,
Jakub

erossignon, 2015-02-14:
Hi Jakub,
I have tried to browserify node-opcua without much success so far as well..
Any joined effort on this would be appreciated...

I have found two major issues:
  - some of the "fs" sync functuions have no direct counterpart in their browserified version
  - the recent introduction of the require("requirish") package, which is used to remove some of the ugliness of the mutliple requires in the code seems to defeat browserify.

erossignon, 2015-06-24:
experimental : Here is how I managed to browserify the simple server on linux
gist:e81eaf4ba1390bf76d2b
Running this scripts produces a single minified JS file 1.9M size ( compressible to 340kb), containing all the sources + dependencies.

dchavesgradiant, 2015-10-01:
Hello,

I managed to browserify a test client on linux using the following commands (after installing all the dependencies with npm):

``` $ browserify bin/client.js -r fs:browserify-fs --exclude memcpy -o test.js```

However, this fails to load in the browser because node-opcua libraries extensively use synchronous filesystem calls (fs.existsSync, . fs.readFileSync, etc.). 

Is there a method to solve this issue and use node-opcua in a browser?

erossignon, 2015-10-02:
have a look at https://github.com/node-opcua/node-opcua/blob/master/brfy.bat
This script is used to browserify and minify the ```simple_server```

```browserify  --bare bin/simple_server.js --exclude usage --exclude memcpy --exclude node-expat --exclude ursa --exclude x509   | minifyjs --mangle toplevel --reserved '$,require,exports'  --comments  --preamble "// * Copyright 2014-2015 NodeOPCUA" -m -o dist/simple_server.min.js```

closed 2016-07-18 by erossignon

fillco, 2018-09-25:
Hello, 
Do you have any progress with running opcua-client in browser? I am currently trying to run node-opcua in react-native and met same issues with 'fs'.

erossignon, 2018-09-25:
node-opcua is still not intended to work inside browser at this time..

zemirco, 2020-03-09:
Hey,

I made our OPC UA client for the browser open source https://github.com/HBM/opcua. It is work in progress but activating a session already works. Maybe it's an option for you.

Best regards,
Mirco