node-opcua/node-opcua #57: NodeCrawler.read() triggers RangeError

Jochen1980, 2015-02-16:
Hi, if I am running NodeCrawler.read() I am getting an error, here is the output of my Enide-Console:
```
MessageBuilder : ERROR DETECTED IN event handler 
[RangeError: Maximum call stack size exceeded]
undefined
```

erossignon, 2015-02-16:
I would need more information to be able to diagnose. Do you have any trace ?

Jochen1980, 2015-02-18:
Here we go, my main-file:
```
// *** joba 16022015 ***
// Einbinden der Module
var http			= require("http");
var async			= require("async");
var opcua			= require("node-opcua");
var crawler			= require('./../node_modules/node-opcua/lib/client/node_crawler.js');
var datamodelNodeId = require('./../node_modules/node-opcua/lib/datamodel/nodeid');
var s				= require("./../node_modules/node-opcua/lib/datamodel/structures");

// Verbindung anlegen, lausche auf Port 80 und liefere client.html-Datei aus;
var connect = require('connect'), serveStatic = require('serve-static');
var http = connect().use(serveStatic(__dirname+'./../client')).listen(80);
var io = require('socket.io')(http); // socket.io-Bibliothek

//Initialisierung der Variablen
var session = null;

// ***************************************************************************
// Funktionen
function dbgFunc(param){ // Debug-Funktion 
	console.log("=== dbgFunc() START ===");
	var dbgParam = param;
	console.log(JSON.stringify(dbgParam, null, 4));
	console.log("=== dbgFunc() ENDE ===");
	return 1;
}

// ****************************************************************************
// Zielstruktur 
var rootFolder = null; // ref[0] von 4 Elementen ist ObjectsFolder
var objectsFolder = null; // ref von 4 Elementen (ns=0;i=40|ns=0;i=35|

// ****************************************************************************
// main-Methode
console.log("BP0");
var client		= new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://" + require("os").hostname() + ":48010";
var session = null;
client.connect(endpointUrl, function(err){
	if (err) {
		console.log("Error : trying to connect ... ",err);
	} else { // connected
		console.log("Connected!");
		var userIdentityToken = new s.AnonymousIdentityToken({policyId: "0"});
		client.createSession(userIdentityToken, function(err, ses){
		    if(err){
		    	console.log("Error : trying to create session  ... ", err);
		    	client.disconnect(function(err){ // disconnect
					if (err){ console.log("Error : trying to disconnect ... ",err); 
					} else { console.log(" disconnected! "); }
				});
		    } else { // session available
		    	console.log("Session created!");
		    	session = ses;
		    	
		    	// NodeToBrowse or Crawl
		    	var nodeId = "ObjectsFolder"; // TODO RangeError Maximum call stack size exceeded
		    	// var nodeId = new opcua.NodeId(datamodelNodeId.NodeIdType.GUID, "360273AA-F2B9-4A7F-A5E3-37B7074E2529", 3);
		    	// var nodeId = new opcua.NodeId(datamodelNodeId.NodeIdType.GUID, "0fbc8d4e-9707-46f9-a6ee-fbd5fd14402c", 3);
		    	
		    	// Node Crawler
		    	var nodeCrawler = new crawler.NodeCrawler(session);
		    	nodeCrawler.read(nodeId, function(){
					if(err){
						console.log("Error : trying to read with NodeCrawler ... ", err);
						session.close(function(err){ // close session
							console.log(" session closed");
							client.disconnect(function(err){ // disconnect
								if (err){ console.log("Error : trying to disconnect ... ",err); 
								} else { console.log(" disconnected! "); }
							});
						});
					} else {
						console.log("BP5");
						treeify.asLines(obj, true, true, function (line) {
							console.log(line);
					    });
						var res = obj; // siehe 
						console.log(JSON.stringify(res, null, 4));
						session.close(function(err){ // close session
							console.log(" session closed");
							client.disconnect(function(err){ // disconnect
								if (err){ console.log("Error : trying to disconnect ... ",err); 
								} else { console.log(" disconnected! "); }
							});
						});
					}
		    	});
		    }
		});
		console.log("BP9");
	}
});
console.log("BP1");
```
The method _crawl gets triggered other and other again until the RangeError occurs. What other information can I offer you?

Jochen1980, 2015-02-18:
I got more information for you: 
If I use the NodeCrawler.crawl-Method and add a counter to the NodeCrawler.emit_on_crawled(element). I can see that the value of the counter is 1814. So on my server there should be around 1800 elements to crawl.

Jochen1980, 2015-03-11:
Hi Etienne, 

I debugged my code and this is the corresponding stack trace:
```
async.queue.q.process.next [D:\joba\...\node_modules\async\lib\async.js:806]  
_toString [D:\joba\...\node_modules\async\lib\async.js:32]  
NodeCrawler._resolve_deferred_browse_node [D:\joba\...\node_modules\node-opcua\lib\client\node_crawler.js:345]  
ClientSession.browse [D:\joba\...\node_modules\node-opcua\lib\client\client_session.js:181]  
ClientSession.performMessageTransaction [D:\joba\...\node_modules\node-opcua\lib\client\client_session.js:628]  
modified_callback [D:\joba\...\node_modules\node-opcua\lib\client\client_secure_channel_layer.js:643]  
process_request_callback D:\joba\...\node_modules\node-opcua\lib\client\client_secure_channel_layer.js:154]  
_on_message_received [D:\joba\...\node_modules\node-opcua\lib\client\client_secure_channel_layer.js:210]   
```
It seems that after the last run of all 1831 objects or nodes in the OPC-UA-Server, some core functions of async-module are throwing an exception. Is this information helpful for detecting the error?

erossignon, 2015-03-12:
It appears that some adress space might expose cyclic graph of nodes. This was causing the crawler to go into infinite recursion
945ca8a1e4b8e1450ad86d50f38abefee883eaae fixes this issue.
Jochen, can you confirm ?

kubat, 2015-03-15:
Hi Etienne,

I am seeing this problem with a Kepware server running a configuration with a large number of tags. I am running:

```bash
> node ./bin/simple_client.js --endpoint opc.tcp://10.37.129.3:49320 --crawl
```

... and this occurring within my code as well.

My OPC server is a Kepware KEPServerEX v5.17.495.0. (should be the latest as downloaded from the Kepware website).

The test configuration file for the Kepware server is located [here](http://web.media.mit.edu/~kubat/Test-big.xml.bz2). It's essentially just a couple of channels with about 6k tags in each channel.

Thanks for the tremendous amount of work you've put into this project!

kubat, 2015-03-16:
Hi Etienne,

Alas, 99ceeec didn't seem to fix the problem. The output I'm seeing to the same command is:

```
(node) warning: Recursive process.nextTick detected. This will break in the next version of node. Please use setImmediate for recursive deferral.
(node) warning: Recursive process.nextTick detected. This will break in the next version of node. Please use setImmediate for recursive deferral.
   ... repeated lots of times ...
(node) warning: Recursive process.nextTick detected. This will break in the next version of node. Please use setImmediate for recursive deferral.

RangeError: Maximum call stack size exceeded
```

erossignon, 2015-03-16:
Hi Rony
 I haven't got this behavior with 0.12 (on ubuntu), however I can reproduce it with node 0.10 (on windows).  
 http://stackoverflow.com/a/15349865/406458

kubat, 2015-03-16:
Confirmed that c7ab0f0 fixes this issue for me.

Jochen1980, 2015-03-18:
Confirmed Version 0.45 does not trigger the RangeError! Thanks a lot Etienne!

closed 2015-03-18 by Jochen1980