node-opcua/node-opcua #59: subscription and client_subscription states are not synchronized

zbysiusiekiera, 2015-03-03:
Hello,

I have noticed interesting behaviour when playing with subscriptions. If I create client_subscription on client side it doesn't detect that associated subscription on server side has expired or was terminated. Shouldn't client detect that subscription is no longer valid, when for example I try to install monitored item after subscription on server side has expired?

Kind regards,
Jakub

erossignon, 2015-03-19:
There should not be need for the client to be notified by the server that the subscription has timed out as the client controls the subscription. The subscription shall not timeout providing that the client keeps sending PublishRequests to the server. 
You found a long standing bug whereby the server may drop the connection if the client does not create monitored item within the livetime time interval.

zbysiusiekiera, 2015-03-25:
Thank you for the fix. I have pulled the changes and it's working great as expected.

closed 2015-08-17 by erossignon