node-opcua/node-opcua #60: return null when read many items together.

majj, 2015-03-07:
When I read 270 float items from KEPServerEX 5.9 by calling readVariableValue,  the returned values from index 190 to 270 is null:

-------------------------------------
  { value: null,
    statusCode: undefined,
    sourceTimestamp: Fri Mar 06 2015 14:55:33 GMT+0800 (China Standard Time),
    sourcePicoseconds: 0,
    serverTimestamp: Fri Mar 06 2015 14:55:33 GMT+0800 (China Standard Time),
    serverPicoseconds: 0 }
-------------------------------------
after I modified the data type with 90 Float items and 180 Boolean items, I got all values.

setInterval(read, 100);

Is there any limitation?

erossignon, 2015-03-07:
It is possible that the KEPServerEX has a maximum number of node that can be set by the client.
( please check the setting on the KEPServerEX side)

It would be nice that node-opcua handle this case seemlessly by making multiple read calls when necessary.

majj, 2015-03-07:
Thanks @erossignon, I'll check.
The user hopes it could read all the values in the same time and then he can get the 'time-series' result(reading time as the result time).   but I found if we read one point twice in the same read call, the returned values are different.

erossignon, 2016-03-23:
closing due to 1 year of inactivity

closed 2016-03-23 by erossignon