node-opcua/node-opcua #63: OPCUA Client seems not working

tomaszpupiec, 2015-03-17:
Hi

I made example client according to https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_client.md. It uses server available at https://opcuaservicesforwpf.codeplex.com/. It works ok with release 0.0.41, but with > 0.0.41 when i try to create session i've got an error:  failure  [Error:  End point must exist opc.tcp://localhost:51212]. Any idea what's the problem ? :)

Code of my example client:

```
var opcua = require("node-opcua");
var async = require("async");

var client = new opcua.OPCUAClient();
var endpointUrl = "opc.tcp://localhost:51212";

var the_session, the_subscription;

async.series([
    function(callback)  {
		client.connect(endpointUrl,function (err) {
			if(err) {
				console.log(" cannot connect to endpoint :" , endpointUrl );
			} else {
				console.log("connected !");
			}
			callback(err);
		});
    },

    function(callback) {
		client.createSession( function(err,session) {
			if(!err) {
				the_session = session;
			}
			callback(err);
		});
    },

    function(callback) {
		the_session.browse("RootFolder", function(err,browse_result){
			if(!err) {
				browse_result[0].references.forEach(function(reference) {
					console.log( reference.browseName);
				});
			}
			callback(err);
		});
    },

    function(callback) {
		the_session.readVariableValue("ns=2;s=Robot1_Axis1", function(err,dataValues) {
			if (!err) {
				console.log(" free mem % = " , dataValues);
			}
			callback(err);
		});
    },

    function(callback) {
		the_session.close(function(err){
			if(err) {
				console.log("session closed failed ?");
			}
			callback();
		});
    }

],
function(err) {
    if (err) {
        console.log(" failure ",err);
    } else {
        console.log("done!");
    }
    client.disconnect(function(){});
}) ;
```

erossignon, 2015-03-18:
This  behavior has been added intentionnaly to conform to the OPCUA specification (CreateSession):

>OPCUA Part 4 Version 1.02 $5.4.1 page 12:
>A Client shall be careful when using the information returned from a Discovery Endpoint since it has no security. A Client does this by comparing the information returned from the Discovery Endpoint to the information returned in the CreateSession response.  [...]
>A  Client  shall verify the  HostName  specified in the  Server Certificate  is the same as the  HostName contained in the  endpointUrl  provided in the  EndpointDescription. If there is a difference  then  the Client  shall report the difference and may close the  SecureChannel.

You need to make sure that your end point is written exactly as one of the endpoint uri provided by the server. 
* You  could try to replace ```var endpointUrl = "opc.tcp://localhost:51212";```  with ```var endpointUrl = "opc.tcp://[put Server HostName here]:51212";```
* alternatively you could use the simple_client.js script to discover the endpoints exposed by your server by running.

```
$node bin\simple_client -e opc.tcp://localhost:51212
```
This will output something like
```
connecting to  opc.tcp://localhost:51212
endpoint                                Application URI                    Security Mode  securityPolicyUri                                Type    certificate
--------------------------------------  ---------------------------------  -------------  -----------------------------------------------  ------  -----------
opc.tcp://HOSTNAME:51212/UA/Server      HOSTNAME:NodeOPCUA-Server          NONE           http://opcfoundation.org/UA/SecurityPolicy#None  SERVER  ...

policyId           tokenType    issuedTokenType  issuerEndpointUrl  securityPolicyUri
-----------------  -----------  ---------------  -----------------  --------------------------------------------------------
username_basic256  USERNAME     null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic256
username_basic128  USERNAME     null             null               http://opcfoundation.org/UA/SecurityPolicy#Basic128Rsa15
                   CERTIFICATE  null             null               null
anonymous          ANONYMOUS    null             null               null
```

```opc.tcp://HOSTNAME:51212/UA/Server``` is the urn that a client should use.

( the simple_client  starts by sending a GetEndpointRequest to the server and print then to the console. GetEndpointRequest is part of the Discovery Service set and does'nt require a OPCUA session and and therefore will accept the ```"opc.tcp://localhost:51212"``` use.) 

Thank your for noticing that tutorial shall be updated accordingly to reflect this behavior.

tomaszpupiec, 2015-03-18:
> You could try to replace var endpointUrl = "opc.tcp://localhost:51212"; with var endpointUrl = "opc.tcp://[put Server HostName here]:51212";

I tried this but didn't help. I solved the problem changing https://github.com/node-opcua/node-opcua/blob/master/lib/client/opcua_client.js like this:

line 81: this.endpoint_must_exist = false;
line 128: this.serverUri = this.endpoint.server.applicationUri;

Thanks for answer.

closed 2016-02-02 by erossignon

JeremyShubert, 2018-06-17:
I was able to create a certificate and it does contain the ip however the endpoint is still my PC name.  Any additional steps for the server to use IP address instead of Host PC name?

limengsong, 2019-08-29:
@erossignon  hi
I would like to ask a question, that is, as the client connects to the server, the server needs a username and password and a certificate. How should I configure it, is there a demo?
[image]
How to configure in the code