node-opcua/node-opcua #66: npm test failed because of missing server_cert_1024.pem

ChristianThiel, 2015-04-01:
What I did:
git clone https://github.com/node-opcua/node-opcua.git
npm install
npm test

The last failed:
fs.js:438
  return binding.open(pathModule._makeLong(path), stringToFlags(flags), mode);
                 ^
Error: ENOENT, no such file or directory 'd:\src\git\opcua\node-opcua\certificat
es\server_cert_1024.pem'
    at Object.fs.openSync (fs.js:438:18)
    at Object.fs.readFileSync (fs.js:289:15)
    at Object.readCertificate (d:\src\git\opcua\node-opcua\lib\misc\crypto_utils
.js:62:22)
    at _load_certificate (d:\src\git\opcua\node-opcua\lib\misc\opcua_secure_obje
ct.js:12:26)
    at OPCUAServer.OPCUASecureObject.getCertificate (d:\src\git\opcua\node-opcua
\lib\misc\opcua_secure_object.js:49:29)
    at new OPCUAServer (d:\src\git\opcua\node-opcua\lib\server\opcua_server.js:2
21:27)
    at Suite.<anonymous> (d:\src\git\opcua\node-opcua\test\test_discovery_server
.js:38:18)
    at context.describe.context.context (d:\src\git\opcua\node-opcua\node_module
s\mocha\lib\interfaces\bdd.js:49:10)
    at Object.<anonymous> (d:\src\git\opcua\node-opcua\test\test_discovery_serve
r.js:24:1)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Module.require (module.js:364:17)
    at require (module.js:380:17)
    at d:\src\git\opcua\node-opcua\node_modules\mocha\lib\mocha.js:187:27
    at Array.forEach (native)
    at Mocha.loadFiles (d:\src\git\opcua\node-opcua\node_modules\mocha\lib\mocha
.js:184:14)
    at Mocha.run (d:\src\git\opcua\node-opcua\node_modules\mocha\lib\mocha.js:40
5:31)
    at Object.<anonymous> (d:\src\git\opcua\node-opcua\node_modules\mocha\bin\_m
ocha:405:16)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3
npm ERR! Test failed.  See above for more details.
npm ERR! not ok code 0

erossignon, 2015-04-01:
It looks like 'npm install'  failed to download and install openssl.

the output of npm install shall be

```
..snip..
checking presence of  C:\temp\a\node-opcua\bin\openssl\openssl.exe
openssl seems to be missing and need to be installed
downloading http://indy.fulgan.com/SSL/openssl-1.0.2-x64_86-win64.zip

Finished writing to disk
done ...
deflating  openssl-1.0.2-x64_86-win64.zip
verifying  OK
done
 .. constructing  C:/temp/a/node-opcua/certificates/PKI
 .. constructing  C:/temp/a/node-opcua/certificates/PKI/CA
 .. constructing  C:\temp\a\node-opcua\certificates\PKI\CA\private
 .. constructing  C:\temp\a\node-opcua\certificates\PKI\CA\certs
 .. constructing  C:\temp\a\node-opcua\certificates\PKI\CA\crl
 .. constructing  C:\temp\a\node-opcua\certificates\PKI\CA\conf

Create Certificate Authority (CA)
=================================

 ROOT = C:/temp/a/node-opcua/certificates/PKI/CA

Generate the CA private Key
===========================

    C:\temp\a\node-opcua\bin\openssl\openssl.exe genrsa -out private/cakey.pem 2048
        err Loading 'screen' into random state - done
        err Generating RSA private key, 2048 bit long modulus
        err ..........+++
        err .................................................................................
        err e is 65537 (0x10001)
..snip..
```

Can you confirm that you you can find the file bin\openssl\openssl.exe after running npm install ?

What is the output of ```node --version``` ?
and ```node -e "console.log(process.platform);" , in your case ?

ChristianThiel, 2015-04-13:
Sorry for the delay, outputs of the suggested commands are:
v0.10.31
win32
ia32
x86

Problem was the wrong download for my platform, must be "indy.fulgan.com/SSL/openssl-1.0.2-i386-win32.zip"

Installed it manually and everything seems running.
Thx for your support.

erossignon, 2015-08-19:
Win64/Win32 installation of openssl was sorted out in acc67883e9b9d6537b5317e3fd76d966d57fa880

closed 2015-08-19 by erossignon