node-opcua/node-opcua #67: Client problem on Raspberry Pi

MKetonen, 2015-04-01:
Hi,

I've tried to find a working OPC Client for Raspberry Pi running CODESYS modified Raspbian. This release there is a Codesys runtime running on the Raspbian with OPC UA server http://store.codesys.com/codesys-control-for-raspberry-pi-sl.html.

I've can create a connection to the server instantly through Unified Automation UaExpert on Win64 running in the same network with the Raspberry. But now that I would like to use the Codesys PLC program variables on my Raspbian program I need the client to run on the Raspberry Pi itself...

OK. So after making the chances in opcua_client.js (endpoint_must_exist = false), i can connect to the server. But now that I'm trying to create session an error occures:

pi@raspberrypi ~/sample_client/node_modules/node-opcua/bin $ node interactive_client.js

 Warning : your current installation is using ersatz-node-expat a full javascript version of node-expat
           This version is slightly slower than node-expat but doesn't require a C++ build during installation

           You can get slightly better performance if you install node-expat
              $ npm install node-expat

           Note: this may require some additional packages to be installed on your system
> open opc.tcp://localhost:4840
 open hostname localhost port : 4840
> >> 132
<< 136
>> 69
<< 367
client connected err= undefined
> cs
>> 1127
> << 403
MessageBuilder : ERROR DETECTED IN event handler { [AssertionError: serverCertificate instanceof Buffer]
  name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: 'serverCertificate instanceof Buffer',
  generatedMessage: false }
AssertionError: serverCertificate instanceof Buffer
    at OPCUAClient._activateSession
(/home/pi/sample_client/node_modules/node-opcua/lib/client/opcua_client.js:348:5)
    at /home/pi/sample_client/node_modules/node-opcua/lib/client/opcua_client.js:468:18
    at /home/pi/sample_client/node_modules/node-opcua/lib/client/opcua_client.js:207:9
    at modified_callback
(/home/pi/sample_client/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:667:24)
    at process_request_callback
(/home/pi/sample_client/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:159:5)
    at _on_message_received
(/home/pi/sample_client/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:216:5)
    at emit (events.js:118:17)
    at MessageBuilder._decode_message_body
(/home/pi/sample_client/node_modules/node-opcua/lib/misc/message_builder.js:431:22)
    at MessageBuilderBase._feed_messageChunk
(/home/pi/sample_client/node_modules/node-opcua/lib/misc/message_builder_base.js:185:18)
    at null.<anonymous>
(/home/pi/sample_client/node_modules/node-opcua/lib/misc/message_builder_base.js:47:14)
    at emit (events.js:107:17)
    at PacketAssembler.feed
(/home/pi/sample_client/node_modules/node-opcua/lib/transport/packet_assembler.js:94:14)
    at MessageBuilderBase.feed
(/home/pi/sample_client/node_modules/node-opcua/lib/misc/message_builder_base.js:149:30)
    at ClientSecureChannelLayer._on_receive_message_chunk
(/home/pi/sample_client/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:581:25)
    at null.<anonymous>
(/home/pi/sample_client/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:312:18)
    at emit (events.js:107:17)
    at _on_message_received
(/home/pi/sample_client/node_modules/node-opcua/lib/transport/tcp_transport.js:167:14)
    at null.<anonymous>
(/home/pi/sample_client/node_modules/node-opcua/lib/transport/tcp_transport.js:241:30)
    at emit (events.js:107:17)
    at PacketAssembler.feed
(/home/pi/sample_client/node_modules/node-opcua/lib/transport/packet_assembler.js:94:14)
    at Socket.<anonymous>
(/home/pi/sample_client/node_modules/node-opcua/lib/transport/tcp_transport.js:248:34)
    at Socket.emit (events.js:107:17)
    at readableAddChunk (_stream_readable.js:163:16)
    at Socket.Readable.push (_stream_readable.js:126:10)
    at TCP.onread (net.js:529:20)

Interactive client shows this:

pi@raspberrypi ~/sample_client/node_modules/node-opcua/bin $ node interactive_client.js

 Warning : your current installation is using ersatz-node-expat a full javascript version of node-expat
           This version is slightly slower than node-expat but doesn't require a C++ build during installation

           You can get slightly better performance if you install node-expat
              $ npm install node-expat

           Note: this may require some additional packages to be installed on your system
> open opc.tcp://localhost:4840
 open hostname localhost port : 4840
> >> 132
<< 136
>> 69
<< 367
client connected err= undefined
> getEndpoints
>> 69
> << 367
└─ 0
   ├─ endpointUrl: opc.tcp://192.168.1.84:4840
   ├─ server
   │  ├─ applicationUri: urn:localhost:OPCFoundation:ServerTestTool
   │  ├─ productUri: http://www.3s-software.com
   │  ├─ applicationName
   │  │  ├─ text: CODESYS OPC UA Server
   │  │  └─ locale: en-Us
   │  ├─ applicationType
   │  │  ├─ key: SERVER
   │  │  ├─ value: 0
   │  │  └─ _options
   │  │     └─ ignoreCase: false
   │  ├─ gatewayServerUri:
   │  ├─ discoveryProfileUri
   │  └─ discoveryUrls
   ├─ serverCertificate
   ├─ securityMode
   │  ├─ key: NONE
   │  ├─ value: 1
   │  └─ _options
   │     └─ ignoreCase: false
   ├─ securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
   ├─ userIdentityTokens
   │  └─ 0
   │     ├─ policyId
   │     ├─ tokenType
   │     │  ├─ key: ANONYMOUS
   │     │  ├─ value: 0
   │     │  └─ _options
   │     │     └─ ignoreCase: false
   │     ├─ issuedTokenType
   │     ├─ issuerEndpointUrl
   │     └─ securityPolicyUri
   ├─ transportProfileUri:
http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
   └─ securityLevel: 0

Any idea what could be the problem?

erossignon, 2015-04-02:
What is your version of node ? ```node --version```
What is your version of node-opcua ? 
have you tried with the latest version in git repo using ```git clone ...``` ?

MKetonen, 2015-04-02:
Node v0.12.1

I fetched the node-opcua yesterday with the instructions on Client side:
$ mkdir sample_client
$ cd sample_client
$ npm init                      # create a package.json
$ npm install node-opcua --save
$ npm install async --save

git clone.. No haven't tried it..

erossignon, 2015-04-02:
npm install node-opcua will install a version taht is slightly out of sync with the version on github
You could try to use
$npm install git+https://github.com/node-opcua/node-opcua.git  --save
in place of 
$ npm install node-opcua --save
in your sample. This will sync node-opcua with the latest version in github regardless of whether it as been packaged to npm or not.

MKetonen, 2015-04-02:
Hi, I just exucuted the $npm install git+https://github.com/node-opcua/node-opcua.git --save on the system previously installed, but it didn't help. The exact same error messages came to screen. I'll try to re-install the whole Raspbian and do a fresh install..

MKetonen, 2015-04-02:
With fresh install problem continues. Did the following:
- fresh microSD with image from Codesys
- sudo apt-get update
- installed node.js 0.12.1 through wget & dpkg
- installed node-opcua through npm (github)
- installed async

And the same error messages on screen. Am I doing something wrong here?

erossignon, 2015-04-02:
OK I think that the [sample tutorial](https://github.com/node-opcua/node-opcua/blob/master/documentation/creating_a_client.md) need to be adapted so that client certificate gets created. Let me do a fresh run on the tutorial on my side to see what's going on

erossignon, 2015-04-02:
Can you tell me the ouptut that the following command provides on your side ?
```$ node node_modules/node-opcua/bin/crypto_create_CA.js```

MKetonen, 2015-04-02:
pi@raspberrypi ~/sample_client/node_modules/node-opcua/bin $ node crypto_create_CA.js
        stdout /usr/bin/openssl
CA private key already exists ... skipping

Create  Application Certificate for Server its private key
==========================================================

 urn =  urn:raspberrypi:NodeOPCUA-Client
private key  /home/pi/sample_client/node_modules/node-opcua/certificates/client_key_1024.pem  already exists
    openssl rsa -pubout -in /home/pi/sample_client/node_modules/node-opcua/certificates/client_key_1024.pem > /home/pi/sample_client/node_modules/node-opcua/certificates/client_public_key_1024.pub
        err writing RSA key
private key  /home/pi/sample_client/node_modules/node-opcua/certificates/client_key_2048.pem  already exists
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/client_cert_1024.pemalready exists => do not overwrite
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/client_cert_2048.pemalready exists => do not overwrite
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/client_selfsigned_cert_1024.pemalready exists => do not overwrite
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/client_selfsigned_cert_2048.pemalready exists => do not overwrite

Create  Application Certificate for Client its private key
==========================================================

 urn =  urn:raspberrypi:NodeOPCUA-Server
private key  /home/pi/sample_client/node_modules/node-opcua/certificates/server_key_1024.pem  already exists
    openssl rsa -pubout -in /home/pi/sample_client/node_modules/node-opcua/certificates/server_key_1024.pem > /home/pi/sample_client/node_modules/node-opcua/certificates/server_public_key_1024.pub
        err writing RSA key
private key  /home/pi/sample_client/node_modules/node-opcua/certificates/server_key_2048.pem  already exists
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/server_cert_1024.pemalready exists => do not overwrite
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/server_cert_2048.pemalready exists => do not overwrite
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/server_selfsigned_cert_1024.pemalready exists => do not overwrite
certificate /home/pi/sample_client/node_modules/node-opcua/certificates/server_selfsigned_cert_2048.pemalready exists => do not overwrite

bastervrugt, 2015-08-14:
I'm experiencing the same problem with codesys opcua server + node-opcua (UaExpert is working)
I've tried multiple workstations (Linux/osx/windows) all with the same result.

I think it has something to do with the serverCertificate that is not present on the codesys opcua server.

https://github.com/node-opcua/node-opcua/blob/master/lib/client/opcua_client.js#L295
```javascript
var serverCertificate = session.serverCertificate; 
assert(serverCertificate instanceof Buffer); 
```

this is the interactive_client.js output after connecting and getEndpoints:
```
> open opc.tcp://192.168.0.50:4840
 open    url : opc.tcp://192.168.0.50:4840
    hostname : 192.168.0.50
        port :  undefined
> client connected :  OK
> getEndpoints
> └─ 0
   ├─ endpointUrl: 
   ├─ server
   │  ├─ applicationUri: urn:localhost:OPCFoundation:ServerTestTool
   │  ├─ productUri: http://www.3s-software.com
   │  ├─ applicationName
   │  │  ├─ text: CODESYS OPC UA Server
   │  │  └─ locale: en-Us
   │  ├─ $applicationType
   │  │  ├─ key: SERVER
   │  │  └─ value: 0
   │  ├─ gatewayServerUri: 
   │  ├─ discoveryProfileUri
   │  └─ discoveryUrls
   ├─ serverCertificate
   ├─ $securityMode
   │  ├─ key: NONE
   │  └─ value: 1
   ├─ securityPolicyUri: http://opcfoundation.org/UA/SecurityPolicy#None
   ├─ userIdentityTokens
   │  └─ 0
   │     ├─ policyId
   │     ├─ $tokenType
   │     │  ├─ key: ANONYMOUS
   │     │  └─ value: 0
   │     ├─ issuedTokenType
   │     ├─ issuerEndpointUrl
   │     └─ securityPolicyUri
   ├─ transportProfileUri: http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
   └─ securityLevel: 0
```
serverCertificate = empty.

and the error trying to create a session:
```
> createSession
> MessageBuilder : ERROR DETECTED IN event handler
{ name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: 'serverCertificate instanceof Buffer' }
AssertionError: serverCertificate instanceof Buffer
    at OPCUAClient._activateSession (/home/ict/Documents/projects/codesys-opcua/node_modules/node-opcua/lib/client/opcua_client.js:340:5)
    at /home/ict/Documents/projects/codesys-opcua/node_modules/node-opcua/lib/client/opcua_client.js:461:18
    at /home/ict/Documents/projects/codesys-opcua/node_modules/node-opcua/lib/client/opcua_client.js:199:9
```

bastervrugt, 2015-08-17:
I can confirm that changing the OPCUAClient.prototype._activateSession function will fix the issue if there's no server certificate present.

find function _activateSession inside \lib\client\opcua_client.js
```javascript
var serverCertificate = session.serverCertificate;
var serverNonce = session.serverNonce;
```
add the following code below
```javascript
if (serverCertificate == null){
  serverCertificate = new Buffer(0);
}
```

sample client will work!
```javascript
the_session.readVariableValue("ns=2;s=|var|Raspberry Pi.application.PLC_PRG.iCounter", function(err,dataValues) {
           if (!err) {
               console.log(" iCounter = " , dataValues);
           }
           callback(err);
       });
```

````
connected !
{ namespaceIndex: 0, name: 'FolderType' }
{ namespaceIndex: 0, name: 'Objects' }
{ namespaceIndex: 0, name: 'Types' }
{ namespaceIndex: 0, name: 'Views' }
 iCounter =  [ { value: { '$dataType': [Object], '$arrayType': [Object], value:
30510 },
    statusCode: { value: 0, description: 'No Error', name: 'Good' },
    sourceTimestamp: null,
    sourcePicoseconds: 0,
    serverTimestamp: null,
    serverPicoseconds: 0 } ]
done!
```

closed 2015-10-01 by erossignon