node-opcua/node-opcua #68: Node.js v 0.12.1

dudanisu, 2015-04-07:
I tried to load server using the tutorial but it gives error;
C:\Program Files\nodejs\node_modules>node
> var opcua = require("node-opcua")
FAST: using node-expat
undefined
C:\Program Files\nodejs\node_modules>node
> var opcua = require("node-opcua")
FAST: using node-expat
undefined
> var server = new opcua.OPCUAServer({ port:4334 });

erossignon, 2015-04-08:
Hi, you probably need to put the command into a file and execute it.
In mytest_server.js
```
var opcua = require("node-opcua")
var server = new opcua.OPCUAServer({ port:4334 });
...
```
(see https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_server.js for instance )

and run it this ways
```node mytest_server.js```

dudanisu, 2015-04-09:
When I load node-opcua on my Ubuntu computer, I get a error npm-debug.log:

0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'install' ]
2 info using npm@1.4.28
3 info using node@v0.10.38
4 verbose node symlink /usr/bin/node
5 verbose config Skipping project config: /home/cs240/.npmrc. (matches userconfig)
6 error install Couldn't read dependencies
7 error package.json ENOENT, open '/home/cs240/package.json'
7 error package.json This is most likely not a problem with npm itself.
7 error package.json npm can't find a package.json file in your current directory.
8 error System Linux 3.2.0-79-generic-pae
9 error command "/usr/bin/node" "/usr/bin/npm" "install"
10 error cwd /home/cs240/node_modules
11 error node -v v0.10.38
12 error npm -v 1.4.28
13 error path /home/cs240/package.json
14 error code ENOPACKAGEJSON
15 error errno 34
16 verbose exit [ 34, true ]

dudanisu, 2015-04-09:
Thanks for replying back.

dudanisu, 2015-04-09:
To install node-opcua I use the command

/home/sudo npm install node-opcua

erossignon, 2015-04-09:
what is /home/sudo ? I guess you mean ```sudo npm install node-opcua``` ?
Before doing ```npm install``  you need to have a valid package.json file in your folder.
I would do :

```
$ mkdir myproject
$ cd myproject
$ npm init
$ npm install node-opcua --save
$ npm install async --save 
```
then edit a script file like sample_server.js inside my project
then
```
$ node sample_server.js
```

dudanisu, 2015-04-09:
Thanks, the server work now and I can use prosys ua client to see the Myvariable1 data.
Thanks again.

dudanisu, 2015-04-09:
When I run the same sample_server.js on windows computer, I get this error:
----------------------------------------------
initialized
Server is now listening ... ( press CTRL+C to stop)
port  4334
 the primary server endpoint url is  opc.tcp://'hostname' is not recognized as a
n internal or external command,
operable program or batch file.:4334/UA/MyLittleServer
------------------------------------------------
Can you please help - the os is windows 8.1.
Thanks

erossignon, 2015-04-10:
It looks like the standard windows command ```hostname.exe``` is missing or is not accessible in your PATH. ```hostname.exe``` returns the name of your computer, this name is used to build the end point url of your server. ```hostname.exe``` can be found under ```c:\Windows\system32\```
Can you test this in a dos console ?
```
C:\>  hostname 
 ```

dudanisu, 2015-04-11:
Thanks, I can now see the opcua server name, however when my client tries to connect I get the error as shown below.
Thanks again.
**********************************************************************************
D:\adopcuaserver>node sample_server.js

 Warning : your current installation is using ersatz-node-expat a full javascrip
t version of node-expat
           This version is slightly slower than node-expat but doesn't require a
 C++ build during installation

           You can get slightly better performance if you install node-expat
              $ npm install node-expat

           Note: this may require some additional packages to be installed on yo
ur system
initialized
Server is now listening ... ( press CTRL+C to stop)
port  4334
 the primary server endpoint url is  opc.tcp://adimadlpt:4334/UA/MyLittleServer
UNSUPPORTED REQUEST !! FindServersRequest
 messages  UNSUPPORTED REQUEST !! FindServersRequest
UNSUPPORTED REQUEST !! FindServersRequest
 messages  UNSUPPORTED REQUEST !! FindServersRequest

erossignon, 2015-04-11:
It looks like your client is considering your server to be a ```discovery server``` rather than a standard server.  ```discovery server```  and its FindServersRequest/FindServersResponse transaction are currently not fully implemented on our end.
Can you tell me more about your client and its configuration ?

erossignon, 2015-08-19:
closing due to lack of activity

dudanisu, 2015-12-22:
Erossignon,
Thanks for your help, I am now again on this project and as u have mentioned I am trying to do step by step. However it give me the error as shown:
D:\myproject> npm install node-opcua --save
npm ERR! Windows_NT 10.0.10240
npm ERR! argv "C:\\Program Files (x86)\\nodejs\\node.exe" "C:\\Program Files (x86)\\nodejs\\node_modules\\npm\\bin\\npm-cli.js" "install" "node-opcua" "--save"
npm ERR! node v5.3.0
npm ERR! npm  v3.3.12
npm ERR! code ENOSELF

npm ERR! Refusing to install node-opcua as a dependency of itself
npm ERR!
npm ERR! If you need help, you may report this error at:
npm ERR!     <https://github.com/npm/npm/issues>

npm ERR! Please include the following file with any support request:
npm ERR!     D:\myproject\npm-debug.log
D:\myproject>

erossignon, 2015-12-22:
@dudanisu it looks like there is something wrong in your d:\myproject folder, like if node-opcua was already installed in D:\myproject2

Could you try this :
 ```
mkdir D:\myproject2
mkdir D:\myproject2\mytest
CD D:\myproject2\mytest
npm init 
npm install node-opcua --save
```

erossignon, 2016-03-13:
closing due to lack of activity.

closed 2016-03-13 by erossignon