node-opcua/node-opcua #73: Server resets sequence number after secure channel renewal

kevinherron, 2015-05-05:
I'm testing your server against a UA benchmarking tool I'm developing and one of the things I've noticed is that the node server is resetting the SequenceNumber in the SequenceHeader of the the message chunk after a secure channel is renewed. This is explicitly now allowed (see Part 6, section 6.2.7), and causes my client to break the connection and reconnect.

Here is a wireshark capture that isolates the exchange: https://dl.dropboxusercontent.com/u/891101/node-opcua-problem.pcapng

erossignon, 2015-08-19:
@kevinherron, can you confirm the reference in the specification ? I cannot find section 6.2.7 in OPCU part 6 ( version 1.0.2 or 1.0.3)

kevinherron, 2015-08-19:
My mistake, I meant Part 6, section 6.7.2.

kevinherron, 2015-08-19:
SequenceNumbers may not be reused for any TokenId. The SecurityToken lifetime should be short enough to ensure that this never happens; however, if it does the receiver should treat it as a transport error and force a reconnect.

The SequenceNumber shall also monotonically increase for all Messages and shall not wrap around until it is greater than 4 294 966 271 (UInt32.MaxValue – 1 024). The first number after the wrap around shall be less than 1 024. **Note that this requirement means that SequenceNumbers do not reset when a new TokenId is issued.** The SequenceNumber shall be incremented by exactly one for each MessageChunk sent unless the communication channel was interrupted and re-established. Gaps are permitted between the SequenceNumber for the last MessageChunk received before the interruption and the SequenceNumber for first MessageChunk received after communication was re- established. Note that the first MessageChunk after a network interruption is always an OpenSecureChannel request or response.

erossignon, 2015-09-18:
@kevinherron can you confirm if the problem still exists in the latest version ?

kevinherron, 2015-09-18:
I followed the example for creating a server [found here](http://node-opcua.github.io/create_a_server.html) and it still doesn't work. I'm unfamiliar with node.js though, and I don't know if following that example ends up using the latest version of the code or not.

erossignon, 2015-09-19:
[image]

erossignon, 2015-09-19:
@kevinherron the easiest way to run a server with the most up to date version:

    $ git clone https://github.com/node-opcua/node-opcua.git
    $ cd node-opcua
    $ npm init
    $ node bin/simple_server

kevinherron, 2015-09-19:
@erossignon Yes, with the latest changes above it is fixed.

closed 2015-09-19 by erossignon