node-opcua/node-opcua #75: UaExpert can't connect

kevinherron, 2015-05-05:
I don't know why, the error reported is Bad_InvalidArgument. It may have something to do with the node server revising the secure channel lifetime to 45 seconds.

Relevant info from UaExpert log:
```
Discovery GetEndpoints on opc.tcp://192.168.184.1:1234 failed (BadInvalidArgument)
[uastack] OpcUa_TcpConnection_Delete: Rude disconnect!
[uastack] ProcessOpenSecureChannelResponse: revised lifetime 60000 below limit 600000.
Discovery FindServers on opc.tcp://192.168.184.1:1234 failed (BadInvalidArgument)
[uastack] OpcUa_TcpConnection_Delete: Rude disconnect!
[uastack] ProcessOpenSecureChannelResponse: revised lifetime 60000 below limit 600000.
```

wngr, 2015-05-12:
Same thing happens to me. Using UaExpert version 1.3.1. Node-opcua 0.0.46

erossignon, 2015-05-12:
I have noticed this as well, For some reason UAExpert rejects the connection abruptly if the client doesn't suggest a timeout lower than 600000.
A fix has been made recently 51cf852858485890afa41b2c99ebc879d5c05a28 to adress this issue.
This will be fix in the next release, in the mean time you can get the fix by cloning the master branch

erossignon, 2015-08-19:
fixed in v0.0.48

closed 2015-08-19 by erossignon