node-opcua/node-opcua #78: sample_server not working on Bodhi Linux and Win7

aw29erac, 2015-06-01:
I am running node v0.12.4, npm v2.11.0 on Bodhi Linux 3.0.0 which is based on Ubuntu 14.04 LTS
when i run 'node sample_server.js' in terminal it says:

 sudo node server
 Warning : your current installation is using ersatz-node-expat a full javascript version of node-expat
           This version is slightly slower than node-expat but doesn't require a C++ build during installation
           You can get slightly better performance if you install node-expat
              $ npm install node-expat
           Note: this may require some additional packages to be installed on your system
initialized
Server is now listening ... ( press CTRL+C to stop)
port  4334
/home/bodhi/hiwi_stuff/server.js:83
        var endpointUrl = server.endpoints[0].endpointDescription().endpointUr
                                              ^
TypeError: undefined is not a function
    at /home/bodhi/hiwi_stuff/server.js:83:47
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:254:17
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:157:25
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:251:21
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:615:34
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/lib/server/opcua_server.js:441:17
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:254:17
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:157:25
    at /home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:251:21
    at OPCUAServerEndPoint._listen_callback (/home/bodhi/hiwi_stuff/node_modules/node-opcua/node_modules/async/lib/async.js:615:34)
    at OPCUAServerEndPoint._end_listen (/home/bodhi/hiwi_stuff/node_modules/node-opcua/lib/server/server_end_point.js:387:10)
    at Server.<anonymous> (/home/bodhi/hiwi_stuff/node_modules/node-opcua/lib/server/server_end_point.js:411:14)
    at Server.g (events.js:199:16)
    at Server.emit (events.js:104:17)
    at net.js:1171:12
    at process._tickCallback (node.js:355:11)

Do you have any idea? Tried it on Windows 7 and got the same issue

erossignon, 2015-06-03:
It looks like the file /home/bodhi/hiwi_stuff/server.js is incomplete. Could you provide it for a check ?

aw29erac, 2015-06-04:
I just copied and pasted the sample_server.js Code from the Server Tutorial.

erossignon, 2015-06-04:
You found a typo. Example need to be updated to reflect recent changes in the API. 
In the mean time you can replace 
```javascript
var endpointUrl = server.endpoints[0].endpointDescription().endpointUrl;
```
with
```javascript
var endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;
```

aw29erac, 2015-06-05:
Thank you very much. It is working now :)

closed 2015-06-22 by aw29erac