node-opcua/node-opcua #80: Connecting with Ignition

cocis, 2015-06-04:
Hi,

I have downloaded the sample server package using git clone and it successfully starts up and listening.

I see your examples of the client code to communicate with the server but I would like to add the server Ignition.(https://inductiveautomation.com/)

I have entered the following settings for the running opcua server as 
IP = <IP address of the Pi>
port 1234
Security Policy "Basic128Rsa15"
Message Security Mode "SignAndEncrypt"
username = "pi"
password = "raspberry"

However the Ignition OPC UA server fails to connect with an error.

Have you tried testing the compliance with ignition before? Or perhaps with other third party tools rather than using the client code published?

Any help you may be able to offer is gratefully received.

erossignon, 2015-06-04:
Does it work with Security NONE ?
What version of node are you using ?. ( ```node --version```) 
If you are using node 0.10 , do not forget to install ```ursa``` (```npm install ursa```

cocis, 2015-06-05:
Hi,

Thank you for the prompt reply It is really appreciated.

node version v0.12.1

Tried with security set to none and get the following fault error in ignition:

com.inductiveautomation.opcua.InternalUAException: Server did not return any endpoints matching the current criteria: SecurityPolicy=Basic128Rsa15, MessageSecurityMode=None, TransportProfileUri=http://opcfoundation.org/UA-Profile/Transport/uatcp-uasc-uabinary
com.inductiveautomation.xopc.client.util.ConnectionUtil.getEndpoint(ConnectionUtil.java:110)
com.inductiveautomation.xopc.client.OpcUaConnection.createUaClient(OpcUaConnection.java:472)
com.inductiveautomation.xopc.client.OpcUaConnection.verifyConnectionState(OpcUaConnection.java:139)
com.inductiveautomation.xopc.client.FailoverAwareOpcUaConnection.verifyConnectionState(FailoverAwareOpcUaConnection.java:75)
com.inductiveautomation.ignition.gateway.opc.OPCManagerImpl$ServerWrapper.verifyConnectionState(OPCManagerImpl.java:745)
com.inductiveautomation.ignition.gateway.opc.OPCManagerImpl$OPCConnectionKeepalive.run(OPCManagerImpl.java:621)
com.inductiveautomation.ignition.common.execution.impl.BasicExecutionEngine$TrackedTask.run(BasicExecutionEngine.java:584)
java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
java.util.concurrent.FutureTask.runAndReset(Unknown Source)
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(Unknown Source)
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
java.lang.Thread.run(Unknown Source)

Attached is a screenshot of the Pi shell showing ipconfig, node version and server output listening.
Also attached is the new ignition settings. If there is any other information you need then please let me know.

[image]
[image]

demeyr, 2015-06-05:
Can you try again while also putting "Security Policy" at "None"?

cocis, 2015-06-05:
Thanks for the reply

I have just tried with both settings set to None and results attached:

[image]

[image]

With these settings I get the error.

com.inductiveautomation.opcua.InternalUAException: java.net.UnknownHostException: darwinPi
com.inductiveautomation.xopc.client.stack.TCPClientChannel.open(TCPClientChannel.java:246)
com.inductiveautomation.xopc.client.stack.UaClient.connect(UaClient.java:66)
com.inductiveautomation.xopc.client.OpcUaConnection.verifyConnectionState(OpcUaConnection.java:140)
com.inductiveautomation.xopc.client.FailoverAwareOpcUaConnection.verifyConnectionState(FailoverAwareOpcUaConnection.java:75)
com.inductiveautomation.ignition.gateway.opc.OPCManagerImpl$ServerWrapper.verifyConnectionState(OPCManagerImpl.java:745)
com.inductiveautomation.ignition.gateway.opc.OPCManagerImpl$OPCConnectionKeepalive.run(OPCManagerImpl.java:621)
com.inductiveautomation.ignition.common.execution.impl.BasicExecutionEngine$TrackedTask.run(BasicExecutionEngine.java:584)
java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
java.util.concurrent.FutureTask.runAndReset(Unknown Source)
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(Unknown Source)
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
java.lang.Thread.run(Unknown Source)
java.net.UnknownHostException: darwinPi
java.net.AbstractPlainSocketImpl.connect(Unknown Source)
java.net.PlainSocketImpl.connect(Unknown Source)
java.net.SocksSocketImpl.connect(Unknown Source)
java.net.Socket.connect(Unknown Source)
com.inductiveautomation.xopc.client.stack.TCPClientChannel.open(TCPClientChannel.java:238)
com.inductiveautomation.xopc.client.stack.UaClient.connect(UaClient.java:66)
com.inductiveautomation.xopc.client.OpcUaConnection.verifyConnectionState(OpcUaConnection.java:140)
com.inductiveautomation.xopc.client.FailoverAwareOpcUaConnection.verifyConnectionState(FailoverAwareOpcUaConnection.java:75)
com.inductiveautomation.ignition.gateway.opc.OPCManagerImpl$ServerWrapper.verifyConnectionState(OPCManagerImpl.java:745)
com.inductiveautomation.ignition.gateway.opc.OPCManagerImpl$OPCConnectionKeepalive.run(OPCManagerImpl.java:621)
com.inductiveautomation.ignition.common.execution.impl.BasicExecutionEngine$TrackedTask.run(BasicExecutionEngine.java:584)
java.util.concurrent.Executors$RunnableAdapter.call(Unknown Source)
java.util.concurrent.FutureTask.runAndReset(Unknown Source)
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(Unknown Source)
java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(Unknown Source)
java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
java.lang.Thread.run(Unknown Source)

Both machines are on the same Network and are able to ping each other.

demeyr, 2015-06-05:
You're connecting to opc.tcp://darwinPi:1234
If your server is configured with a resourcepath then you probably need to connect to something like opc.tcp://darwinPi:1234/UA/MyLittleServer
(see your sample server code for how it's configured exactly)

Not sure how you can configure this in Ignition.
Maybe like this:
host: darwinPi/UA/MyLittleServer
port: 1234

cocis, 2015-06-05:
Thanks for the reply.

I am not exactly sure how node opcua is setting up the server with regard to the resource path. The code used is cloned from the sample.js and shown below:

var opcua = require("node-opcua");
var os = require("os");

var default_xmlFile = __dirname + "/node_modules/node-opcua/nodesets/Opc.Ua.NodeSet2.xml";
console.log(" node set ", default_xmlFile);

// Let create an instance of OPCUAServer
var server = new opcua.OPCUAServer({
    port: 1234,        // the port of the listening socket of the server
    nodeset_filename: default_xmlFile
});

// we can set the buildInfo
server.buildInfo.productName = "MySampleServer1";
server.buildInfo.buildNumber = "7658";
server.buildInfo.buildDate = new Date(2014, 5, 2);

// the server needs to be initialized first. During initialisation,
// the server will construct its default namespace.
server.initialize(function () {

    console.log("initialized");

    // we can now extend the default name space with our variables
    construct_my_address_space(server);

    // we can now start the server
    server.start(function () {
        console.log("Server is now listening ... ( press CTRL+C to stop)");
        var endpointUrl = server.endpoints[0].endpointDescriptions()[0].endpointUrl;

        server.endpoints[0].endpointDescriptions().forEach(function (endpoint) {
            console.log(endpoint.endpointUrl, endpoint.securityMode.toString(), endpoint.securityPolicyUri.toString());
        });
    })

});

function construct_my_address_space(server) {
    // we create a new folder under RootFolder
    server.engine.createFolder("RootFolder", {browseName: "MyDevice"});

    // now let's add first variable in folder
    // the addVariableInFolder
    var variable1 = 10.0;

    server.nodeVariable1 = server.engine.addVariableInFolder("MyDevice", {
        nodeId: "ns=4;b=1020ffaa", // some opaque NodeId in namespace 4
        browseName: "MyVariable1",
        dataType: "Double",
        value: {
            get: function () {
                var t = new Date() / 10000.0;
                var value = variable1 + 10.0 * Math.sin(t);
                return new opcua.Variant({dataType: opcua.DataType.Double, value: value});
            }
        }
    });

    ///
    var variable2 = 10.0;

    server.nodeVariable2 = server.engine.addVariableInFolder("MyDevice", {
        browseName: "MyVariable2",
        dataType: "Double",
        value: {
            get: function () {
                return new opcua.Variant({dataType: opcua.DataType.Double, value: variable2});
            },
            set: function (variant) {
                variable2 = parseFloat(variant.value);
                return opcua.StatusCodes.Good;
            }
        }
    });

    server.nodeVariable3 = server.engine.addVariableInFolder("MyDevice", {
        nodeId: "ns=4;b=1020ffab", // some opaque NodeId in namespace 4
        browseName: "Percentage Memory Used",
        dataType: "Double",
        value: {
            get: function () {
                // var value = process.memoryUsage().heapUsed / 1000000;
                var percentageMemUsed = 1.0 - (os.freemem() / os.totalmem() );
                var value = percentageMemUsed * 100;
                return new opcua.Variant({dataType: opcua.DataType.Double, value: value});
            }
        }
    });

}

When running it simply shows:
opc.tcp://darwinPi:1234 as endpoint.endpointUrl in the console

I therefore assumed that there is no resource path rightly or wrongly.

And I can't see any settings where the server is instantiated:

// Let create an instance of OPCUAServer
var server = new opcua.OPCUAServer({
    port: 1234,        // the port of the listening socket of the server
    nodeset_filename: default_xmlFile
});

Still struggling to get the sample code provided connected to ignition.

demeyr, 2015-06-05:
Indeed that's correct.

How about a  firewall?
Can you try with port 80 instead?

An maybe try this:
https://inductiveautomation.com/forum/viewtopic.php?f=70&p=37080

cocis, 2015-06-05:
Changed to Port 80 

Had to run "sudo node server.js" due to port permissions error

[image]

And still the same issue from ignition.

As i have cloned the sample straight from the site could the certificates be anything to do with this as I am using those downloaded? That may be a dumb question.

cocis, 2015-06-05:
Boom! 

demeyr you are a genius and my saviour.

I just tried the solution you linked with the host overide and it has connected.

I owe you a beer.

Here are the settings that worked. 

[image]

cocis, 2015-06-05:
I have one final issue:

The code appears to define a Root folder MyDevice and adds to this the "MyVariable1", "MyVariable2" and "MyVariable3" function.

MyVariable 1 & 3 hve getter functions only and "MyVariable3" has a setter and getter.

In the Ignition quick client I can access the build information but I dont see "My Device" and the 3 Variables.

See screenshot:

[image]

cocis, 2015-06-05:
I found this link 
https://inductiveautomation.com/forum/viewtopic.php?f=72&t=13894&p=50650&hilit=node+opc&sid=f8feee78a6c3c5bcee313173464dce61#p50650

It states:
"You know, I just looked at the node-opcua server example and it looks like it adds the folder and nodes to the root of the address space. Ignition's OPC-UA client doesn't browse from root, it browses from the Objects folder. 

If you modify the example to add things starting from the Objects folder you'll probably see the nodes you're looking for."

Example Code is:

server.engine.createFolder("RootFolder", {browseName: "MyDevice"});

I simply changed "RootFolder" to "ObjectsFolder" and now it works as expected.

Thank you for all the help.

UmerIftikhar, 2016-03-10:
java.lang.IllegalArgumentException
Could some one plz help me regarding this, I have my node-server running, but its also configured in the ignition   OPC Connections -> Server.
But when I access it from the OPC Connections -> Quick Client, I get this error.
[image]

alamsha, 2016-03-10:
First test your OPC UA server with UaExpert OPC UA cleint and Prosys OPC UA client. Use these 2 tools to troubleshoot the behaviour of any other client.

1. UaExpert v1.4.1
https://www.unified-automation.com/downloads/opc-ua-clients.html

2. Prosys OPC UA Client
https://www.prosysopc.com/products/opc-ua-client/

3. You can access my Ignition server at:
opc.tcp://alamscada.dynu.com:4096

erossignon, 2016-03-11:
see #165 (@InsideManCulver)
> The problem is within Ignition version. Ignition version 7.8.1 gives the java error. In the version 7.8.2 this bug is fixed so updating to this version fixes the problem.

closed 2016-07-18 by erossignon