node-opcua/node-opcua #81: Historical and A&E access on client side

demeyr, 2015-06-05:
Could you use help with adding the Historical and A&E access on the client?

We have successfully adjusted the code to request Historical values.
(Tested with Prosys Historian)
It's working because we can see the correct answers in wireshark.

Any tips or guidelines on where to start adjusting Node OPC UA for processing the answers?

erossignon, 2015-06-11:
We shall first of all create standard HA strutcure that are missing

demeyr, 2015-06-12:
I see you already created the structures.

I created a pull-request for adding the readHistoryValue() function in opcua_client.js

Once this is added it should work, at least for basic reads.

We tested successfully with the trial version of Prosys Historian.
The tags inside Prosys are coming from a trial version of KEPServerEX.

The disadvantage of this setup is that it's a bit complicated for testing + A&E is not possible due to Kepware only exposing DA over OPC UA towards the Prosys Historian.

A solution could be to test with the Prosys Simulation Server instead.

closed 2018-01-25 by erossignon