node-opcua/node-opcua #82: Array Types

cocis, 2015-06-07:
Hi this may be a simple question but I am struggling to resolve the issue.

I have in my code a file that has been written as a JSON dump from Python.

The file i.e. team.json contains several rows with each row holding the first name, last name and role.

In node.js i read the file as:

var teamObj = JSON.parse('fs.readFileSync('/home/pi/files/team.json', 'utf8'))

I now need to make the contents of teamObj available within ignition from the opcua server;

   server.nodeVariable5 = server.engine.addVariableInFolder ("MyDevice", {
        nodeId: "ns=4;b=1020ffad", // some opaque NodeId in namespace 4
        browseName: "Team List",
        dataType: "String",
        valueRank: 1,
        value: {
            get: function () {
                                  return new opcua.Variant({dataType: opcua.DataType.String, arrayType:variantArrayType.Array, value: teamObj });
            }
         }
   });

The above is probably wildly wrong and for that I apologise. 
Do you have an example of a getter for an array of objects read in such a manner?

erossignon, 2015-06-07:
```teamObj```  is a javascript object with properties, and not a array of string.
could you send a sample ```team.json``` ? 

if you create teamObj this way it should work:
```
teamObj = ["team1","team2","team3"];
```

cocis, 2015-06-07:
Thank you for the reply.
Here is a simple example of the team.json (attached)

The file is written by some Python code and I am reading this file in the node.js code as a crude way of interfacing.
[image]

It reads in great and I can access the the object as teamObj[0].fname, etc but i then need to make the whole object structure available in Ignition via the OPC UA  server.

I'm sorry to trouble you with this but all the examples I saw seem to use the same DataType "Double".

erossignon, 2015-06-08:
You could create a "Team" folder and expose each team member in a dedicated object with 3 properties each.

```javascript
    var list = [
        { fname:"John",  lname:"Doe" , role: "Manager"},
        { fname:"Bob",   lname:"Doe" , role: "Technician"}
        { fname:"Garry", lname:"Doe" , role: "Operator"}
    ];

    function createList(list ) {

        var index = 0;
        function addPerson (parentFolder,person) {

            index+=1;

            var obj = server.engine.createFolder(parentFolder, { browseName: "Member-"+ index.toString() });
            server.engine.addVariable(obj,{ browseName:"ForName",  dataType: "String", value: { dataType: DataType.String, value: person.fname}});
            server.engine.addVariable(obj,{ browseName:"LastName", dataType: "String", value: { dataType: DataType.String, value: person.lastName}});
            server.engine.addVariable(obj,{ browseName:"Role",     dataType: "String", value: { dataType: DataType.String, value: person.role}});
        }
        var team = server.engine.createFolder("ObjectsFolder", { browseName: "Team"});
        list.forEach(addPerson.bind(null,team));

    }
    createList(list);
```
[image]

cocis, 2015-06-08:
Thank you so much for the great example, I really appreciate it.

Hats off to you Sir.

cocis, 2015-06-09:
One slight problem I have is if the list is changing:

I can attach a get function in the for loop but on each pass the value of the get needs to be assigned to a specific value of the object array.

i.e. get: function () {
             return new.opcua.Variant({dataType: opcua.DataType.String, value:  list[0].fname })
      }

I have tried using list[index-1] within the foreach loop but this just returns the same values from the last row of all entries as index remains at 3 after the creation.

Within the loop I therefore need to lock each iteration to the external list array entry which will be changing.

Another thought is the possibility of calling createList(list) again after initialisation.

Can you offer any guidance?

cocis, 2015-06-09:
I still have a slight problem if the "list" object is changing and I need to retrieve continuously updated values.

If I add a "get" function for each "addVariable" what should I set the "value :" equal to in order to lock it to a particular row of the list? i.e. list[0].fname etc

Secondly, if the list is dynamic and it is read again and now only has 2 entries as opposed to 3, i'm not sure how to handle this in the code. Can I dynamically remove and add objects exposed by the opcua server or do I need to maintain a max list size?

Apologies once more for the dumb questions.

erossignon, 2015-06-10:
Objects  in opcua are identified with nodeId and therefore must have stability and represent the same thing.
There is probably not enough in your example to detect if a change in the list entries correspond to a new object being added and the previous one being deleted or if this change correspond to the change of propertie of the object at the specified position in the line.
What is the key value that determine the uniqueness of a value in the list ?
In your case; what is the strategy you would use to reassociate objects from the external list and existing OPCUA objects ?

cocis, 2015-06-10:
Thank you for the response and apologies for lack of clarity.

I have the following JSON format which let us say represents the staff currently rostered:

    var list = [
        { fname:"John",  lname:"Doe" , role: "Manager"},
        { fname:"Bob",   lname:"Doe" , role: "Technician"}
        { fname:"Garry", lname:"Doe" , role: "Operator"}
    ];

Naturally this list will change and there may be more staff on duty or less at a given time of day.
The minimum number will be 0 at plant closing time and the maximum will be 5 during the peak period.

There is a separate process that writes out the JSON  "staff list" to a file entitled "stafflist.json"

The node.js reads this file with the following callback code.
------------------------------------------------------------------------------------------------
var fs = require('fs');

function  getStaff() { 
   staffObj = JSON.parse(fs.readFileSync('/home/pi/stafflist.json','utf8'));
   // console.log( staffObj[0].fname );
   return (staffObj );
}

// Refresh staff info  from roster every second
setInterval ( function() {
      staffObj = getStaff();
       console.log("First Name: " + staffObj[0].fname + "   Last Name: " +staffObj[0].lname);
},5000);

So as you can see every 5 seconds this staffObj is changing. Both the properties and quantity of staff objects may change. Let us assume the role they perform in the shift can change as frequently as every 5 seconds. So the next callback of the interval function could result in new staff, additional staff, or the same staff but the "role" property has changed.

The neat code example you provided works perfectly on call of the createList() function, but it cannot manage the changes that may happen subsequently as a result of the setInterval function.

I amended your example for each server.engine.addVariable.... statement to include a get: function as follows:

server.engine.addVariable(obj, { 
                                             browseName: "role",
                                             dataType: "String",
                                             value: {
                                                  get: function () {
                                                      return new opcua.Variant({  dataType: opcua.DataType.String, value: staffObj[index].role })
                                                  }
                                             }

This populates the list ok at startup but of course changes all objects to the properties of the last object as index was left pointing to the last object when the foreach loop executed.

I have therefore had to remove the foreach loop (cry) and explicitly state the array index for all staff objects read by changing the reurn statement  to read:

 return new opcua.Variant({  dataType: opcua.DataType.String, value: staffObj[0].role })

This binds it to the first entry in the staffObj at the particular index and the role updates as expected.

It doesn't however help me when the number of staff on duty reduces to zero and an exception is thrown in the code as I try and access and index that  no longer exists.

Please let me know if you need any further information. Fingers crossed you can help me because I am really stuck with this at the moment. All I can think of doing is padding the JSON file with blanks to prevent this (Keeping the contents of the file and hence the dimensions of staffObj fixed.

I'm not sure if this is the correct way of doing it and it probably qualifies as a hack.

This is because the value: person.role does not change when the get function is called.

erossignon, 2015-06-11:
Regarding the access violation.
If the data is not available, it is valid to return a StatusCode rather than a Variant on the get function
```
if (index < staffObj.length) {
    return new opcua.Variant({  dataType: opcua.DataType.String, value: staffObj[index].role })
} else {
   return  opcua.StatusCode.BadDataUnavailable;
}
```

I believe that you would also need the ability to remove already inserted node from the adress space. At this time this is not yet implemented. => Let see what we can do.

cocis, 2015-06-11:
Thank you for the reply and all the hard work done to date.

closed 2016-03-13 by erossignon

Zaniyar, 2018-09-10:
@erossignon is server.engine.createFolder() removed from the actual version? How can we still create folders?