node-opcua/node-opcua #85: How to detect lost connection to OPC server

kgmoney, 2015-06-12:
Hi,

Awesome project by the way!  I'm using this to connect/monitor OPC tags in Kepserver.  Everything is working great, except if I loose my connection the OPC server (network outage for example).  Is there an event or some property that will tell me the status of my connection?  I've tried all the subscription events (internal_error and status_changed looked especially interesting) but none of them fire when I break the network connection.

Thanks for your help,
Kevin

demeyr, 2015-06-16:
There's probably a better solution but in the meantime you could solve it by monitoring a watchdog.

kgmoney, 2015-06-16:
Thanks for the suggestion.  That kinda what I ended up with...  On the subscription events 'keepalive' and 'received_notification' I'm updating a last updated time and then checking that time periodically to see if the updates have stopped.

I think the correct way has something to do with the following (http://node-opcua.github.io/api_doc/classes/ClientSubscription.html), but I couldn't get it to timeout the connection:

- requestedPublishingInterval
- requestedLifetimeCount
- requestedMaxKeepAliveCount

erossignon, 2015-06-22:
In fact the underlying socket  is quite resilient and doesn't report wire disconnection very easily.
We  need to implement some better mechanism to detect timeout on the client side.

closed 2015-08-19 by erossignon