node-opcua/node-opcua #86: Create TypeDefinition

martinergb, 2015-06-18:
Hi, Am I able to create a new TypeDefinition and then instantiate it?

erossignon, 2015-06-19:
Yes, it's possible , but the API is not that friendly yet.
Can you provide a example of the type you've liked to create ?  This will help us to provide a dedicated sample and enhance the API, with practical example in mind.

martinergb, 2015-06-19:
For convenience, can you take the example in Chapter 3 of the book "OPC Unified Architecture" by Wolfgang Mahnke? Create an AirConditionerController and then instantiate 3 controllers from it.
Thank you.

erossignon, 2015-06-20:
![](http://yuml.me/diagram/scruffy/class/[BaseObjectType{bg:green}], [BaseDataVariabeType{bg:green}], [BaseVariabeType{bg:green}], [TemperatureSensorType{bg:blue}], [MachineType{bg:blue}], [HeatherSwitch{bg:yellow}], [Temperature{bg:yellow}], [Temperature{bg:yellow}], , [BaseVariabeType]^[BaseDataVariabeType], , [BaseObjectType]^[TemperatureSensorType], [BaseObjectType]^[MachineType], , [BaseDataVariabeType]^[Temperature], [BaseDataVariabeType]^[HeatherSwitch], , [MachineType] ++-- [TemperatureSensor], [MachineType] ++-- [HeatherSwitch], [MachineType] ++-- [SetSwitchState{bg:orange}], , [TemperatureSensorType]^[TemperatureSensor], [TemperatureSensorType]++--[Temperature].)

martinergb, 2015-06-21:
Can you provide some code using node-opcua to represent this UML diagram?

mpostol, 2015-06-21:
Hi,

Acording to the specification, i.e. Part 6 Mapping the hope is that the Address Space is created using xml files compliant with the NodeSet schema ([Address Space Interchange XML](http://goo.gl/LE64MA)), but not using API. 
More about the Information Model and Address Space you can learn from my articles pointed out in the description of the project: [OPC Unified Architecture - Object Oriented Internet](https://github.com/mpostol/OPC-UA-OOI)

One purpose of this project is to develop methodology and tools supporting publication of any kind of data in the context of OPC UA Address Space as the meta-data without programing.
In the white-paper [OPC UA Information Model Deployment] (http://goo.gl/HqYjvy)  you can find description of two models:

* Boiler (the most popular as far as I know) and 
* ADI (the most complex as far as I know). 

The models are attached to the tool CAS Address Space Model Designer available for free from here:

[OPC UA Address Space Model Designer](http://www.commsvr.com/Products/OPCUA/UAModelDesigner.aspx)

As the member of a few OPCF Working Group (UA, CMP, Pub-sub) I am willing helper to support this project But as I am not familiar with Java I cannot contribute in the code development.

Best Regards,
Mariusz

erossignon, 2015-06-24:
Mariuz,
Node-opcua can now aggreate more than one nodeset files at startup to create the initial address space as recommanded. Sometime it is necessary to be able to create new type instance on the fly and a good, easy to use api is necessary for this.

erossignon, 2015-06-24:
@martinergb 
Here is a file that show you how to create the TemperatureSensorType and MachineType.
As you will see , in this initial version ,the code is quite convoluted and instanciation has to be made manually. The code will evolve as the API get clearer, we shall also implement a class instanciation facility. Stay tune

martinergb, 2015-06-25:
Thank you @mpostol  and @erossignon  for your information

In the function instantiateTemperatureSensor, I think you forget to add Temperature Property to temperatureSensorTypeNode. Is that correct?

mpostol, 2015-06-25:
@erossignon - my concern is that adoption of the OPC UA by the end user requires "deployment without programming". Main goal of the @OPC-UA-OOI project it to work on GUI instead API. It is good news as we could converge both projects in spite they are based on different programming language platforms. By the way I have a question about your plans relating to Information Model representation (ModelDesign, something else) and development of a compiler generating NodeSet xml files.

closed 2016-10-28 by erossignon