node-opcua/node-opcua #87: Interop issue with UALDS 1.02 (BadSecurityPolicyRejected)

eoursel, 2015-06-20:
Tested on Windows 10 Preview and UALDS 1.02.335.0

Looks like registration on the UALDS is not working

node .\bin\simple_server.js

 Warning : your current installation is using ersatz-node-expat a full javascrip
t version of node-expat
           This version is slightly slower than node-expat but doesn't require a
 C++ build during installation

           You can get slightly better performance if you install node-expat
              $ npm install node-expat

           Note: this may require some additional packages to be installed on yo
ur system

registering server to :opc.tcp://zion:4840/UADiscovery
  server PID          : 16884
 ERROR RECEIVED FROM SENDER BadSecurityPolicyRejected (0x80550000) null
00000000: 45 52 52 46 10 00 00 00 00 00 55 80 ff ff ff ff
                             ERRF......U.....

MESSAGE BUILDER ERROR ERROR RECEIVED
request id =  2
 message was
undefined
 message was 2: { /*RegisterServerRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;i=0
   timeStamp                   /* UtcTime                          */: 2015-06-2
0T13:53:57.564Z
   requestHandle               /* IntegerId                        */: 3
       0x3
   returnDiagnostics           /* UInt32                           */: 0
       0x0
   auditEntryId                /* UAString                         */: null
   timeoutHint                 /* UInt32                           */: 0
       0x0
   additionalHeader            /* ExtensionObject                  */: null
 }
 server                        /* RegisteredServer                 */: {
   serverUri                   /* String                           */: request.s
erverUri
   productUri                  /* String                           */: request.p
roductUri
   serverNames                 /* LocalizedText                 [] */: [
     { /*0*/
       text                    /* String                           */: MyServerN
ame
       locale                  /* LocaleId                         */: en
     }
   ]
   serverType                  /* ApplicationType                  */: SERVER (
0)
   gatewayServerUri            /* String                           */: null
   discoveryUrls               /* String                        [] */: [ ] (l=0)

   semaphoreFilePath           /* String                           */: null
   isOnline                    /* Boolean                          */: null
 }
};

D:\Users\Eric\Git\Github\node-opcua\node_modules\better-assert\index.js:37
  throw err;
        ^
AssertionError: response instanceof RegisterServerResponse
    at D:\Users\Eric\Git\Github\node-opcua\lib\server\opcua_server.js:2059:17
    at modified_callback (D:\Users\Eric\Git\Github\node-opcua\lib\client\client_
secure_channel_layer.js:716:24)
    at process_request_callback (D:\Users\Eric\Git\Github\node-opcua\lib\client\
client_secure_channel_layer.js:70:5)
    at D:\Users\Eric\Git\Github\node-opcua\lib\client\client_secure_channel_laye
r.js:284:9
    at Array.forEach (native)
    at _cancel_pending_transactions (D:\Users\Eric\Git\Github\node-opcua\lib\cli
ent\client_secure_channel_layer.js:280:37)
    at _on_transport_closed (D:\Users\Eric\Git\Github\node-opcua\lib\client\clie
nt_secure_channel_layer.js:309:34)
    at emit (events.js:107:17)
    at _on_socket_ended_message (D:\Users\Eric\Git\Github\node-opcua\lib\transpo
rt\tcp_transport.js:219:10)
    at Socket.<anonymous> (D:\Users\Eric\Git\Github\node-opcua\lib\transport\tcp
_transport.js:263:34)
    at Socket.emit (events.js:129:20)
    at _stream_readable.js:908:16
    at process._tickCallback (node.js:355:11)
PS D:\Users\Eric\Git\Github\node-opcua>

eoursel, 2015-06-20:
Have to modify the configuration of the UALDS to enable SecurityMode None... but there is still an issue at least with the sample server found in the trace file of the UALDS. 

Sat Jun 20 16:11:09 [9980]: ServerUri 'request.serverUri' is invalid (needs to start with 'urn:' or 'http://').

erossignon, 2016-03-13:
Tested, now working! 
 ```node .\bin\simple_server.js``` can register server to the Local Discovery Server at endpoint opc.tcp://localhost:4840/UADiscovery .

closed 2016-03-13 by erossignon

wcc21320, 2016-11-23:
In 0.0.60, Run the test case "test/descovery/test_discovery_server.js";
Always has BadSecurityPolicyRejected :

error   ServiceFault returned by server { /*ServiceFault*/
 responseHeader                /* ResponseHeader                   */: {
   timeStamp                   /* UtcTime                          */: 2016-11-23T03:40:40.963Z
   requestHandle               /* IntegerId                        */: 1               0x1
   serviceResult               /* StatusCode                       */: BadSecurityPolicyRejected (0x80550000)
   serviceDiagnostics          /* DiagnosticInfo                   */: {
     namespaceUri              /* Int32                            */: -1
     symbolicId                /* Int32                            */: -1
     locale                    /* Int32                            */: -1
     localizedText             /* Int32                            */: -1
     additionalInfo            /* String                           */: null
     innerStatusCode           /* StatusCode                       */: Good (0x00000)
     innerDiagnosticInfo       /* DiagnosticInfo                   */: {
     }
   }
   stringTable                 /* String                        [] */: [ ] (l=1)
   additionalHeader            /* ExtensionObject                  */: null
 }
};

Any suggestion will be nice, thanks.