node-opcua/node-opcua #88: Interop issue with OPC Foundation standard Client 1.02.335

eoursel, 2015-06-22:
"Server not responding to keep alive requests"

So the client automatically disconnect the server

erossignon, 2015-08-30:
Why is that ?
the client seems to be reopening new channels over and over again, what's going on ?

erossignon, 2015-08-30:
I ran the debug version of the OPCFoundation standard client and got an interesting debug log:
```
5852 - 21:53:50.716 Unexpected keep alive error occurred: The server returned data value of type UInt32 when a value of type Int32 was expected.```

In order to initiate a keep alive request, the client sent a ReadRequest on i=2259 ( Server_ServerStatus_State). 
node-opcua replies this request by returning  the state enum value as a UInt32 variant.
It seems that this generates a exception inside the client as a Int32 variant is expected instead.

erossignon, 2015-08-30:
OPCUA Specification 1.0.3 part 3 : 5.8.2
[...]Enumeration are always encoded as Int32 on the wire[...]

closed 2015-09-19 by erossignon