node-opcua/node-opcua #91: Write, writeAttribute, writeSingleNode etc. are not changing the values of tags in the address space.

Orobas, 2015-07-16:
I am attempting to use node-opcua to host a server which exposes several tags for real time data monitoring. I currently have the JSON data being passed through a socket to the server, which then tries to write to and change the values of the tags. If I read the tags, it looks like it changes successfully but when I read the data from my client or UAExpert, the data always stays the same as when it was initially set. No error statuscodes are being triggered from the write functions. I'm not sure if it has something to do with how I set up my address space. My address space is set up through a file AddrSpace.js which has 3 functions, one to make the initial address space, an addition function which adds a branch onto the tree with a computer name, and an update function which is where the values are supposed to be updated. 

Add_Address_Space
```javascript
Add_Address_Space : function(server, Compname,record,create) {
           if(create) {
server.Compname.Test.ComputerName = server.engine.addVariable(server.Compname.Test, {
                nodeId: "ns=5;s=" + Compname + ".Test.ComputerName",
                browseName: "ComputerName",
                dataType: "String",
                value: {
                    get: function () {
                        return new opcua.Variant({dataType: opcua.DataType.String, value: record.COMPUTER_NAME});
                    }
                }
            });
       }
};
```

Update_Address_Space
```javascript
Update_Address_Space : function(server,Compname,obj){
var nodesToWrite =[new WriteValue({
            nodeId: "ns=5;s="+Compname+".Test.ComputerName",
            attributeId: 13,
            value: {
                value: {
                    dataType: opcua.DataType.String, value: obj.COMPUTER_NAME
                }
            }...rest of nodes...] ;
server.engine.write(nodesToWrite,function(err,statusCode){
            if (err) console.log(err);
        });
```
The code that tells the tag to update is here...
```javascript
addspace.Update_Address_Space(server,computername,obj);
console.log("EXPECTED: "+obj.COMPUTER_NAME);
console.log("ACTUAL:   "+server.engine.readSingleNode("ns=5;s="+computername+".Test.ComputerName",13).value.value);
```
which will return that the expected and the actual are the same, indicating that the values should of changed.

EXPECTED: DOMAINNAME725
ACTUAL:   DOMAINNAME725

But when I call 
```javascript
var nodesToRead =[{
    nodeId: "ns=5;s=CLCONC4686.Test.ComputerName",
    attributeId: 13
}...rest of nodes...];
the_session.read(nodesToRead,function(err,nodesToRead,results,diagnosticInfos) {
            if (!err) {
                console.log( " SECOND TEST =" , results[0].value.value);
            }
        });
```
it returns
 SECOND TEST = DOMAINNAME1

Here's a picture showing that the address space is actually working except for the value changing part with a screenshot from UAExpert, showing the value of ComputerName. 
[image]

If you have any suggestions or fixes please let me know, I've been trying a bunch of different workarounds to no avail right now. Thanks in advance, and keep up the good work on this project.

erossignon, 2015-07-17:
You can get more information about what's going on by printing the statusCode returns by the read/write operation

```javascript
server.engine.write(nodesToWrite,function(err,statusCode){
            if (err) { console.log(err);}
            console.log("write statusCode ", statusCode);
        });
...
the_session.read(nodesToRead,function(err,nodesToRead,results,diagnosticInfos) {
            if (!err) {
                console.log( " SECOND TEST =" , results[0].toString());
            }
        });
```

I see a few problems here:
* you are using the get/set form to access  the underlying Variant when constructing the variable.
  the get function is provided but there is no set  function, as such your variable behalves like   readonly.
 * you should probably use the simple form without get/set :
``` javascript
server.Compname.Test.ComputerName = server.engine.addVariable(server.Compname.Test, {
                nodeId: "ns=5;s=" + Compname + ".Test.ComputerName",
                browseName: "ComputerName",
                dataType: "String",
                value:  new opcua.Variant({dataType: opcua.DataType.String, value: record.COMPUTER_NAME});
                }
            });
       }
```
or provide a set function ; see example in https://github.com/node-opcua/node-opcua/blob/master/test/helpers/build_server_with_temperature_device.js#L125

Orobas, 2015-07-17:
Thank you for the reply. Getting rid of the get worked perfectly. Your suggestion for the statusCode didn't help as it would return Good (0x00000), even though it did not work correctly.

closed 2015-07-17 by Orobas