node-opcua/node-opcua #92: browse subfolders

aw29erac, 2015-07-17:
Hi,
this might sound easy, but how do i browse subfolders? I'm searching for hours by now.
Btw. thanks for your great work!

zombinary, 2015-07-17:
Hi aw29erac,
in my experience you can not browse a sub-folder automatically.
You must browse the parent folder, 
this returns the NamespaceIndex and Identifier from the sub-folder.
Then you must browsed again.

aw29erac, 2015-07-17:
thanks. how do i read the NamespaceIndex and Identifier?

zombinary, 2015-07-17:
Here is an example to browse the opc-ua "root" but it is not tested. I hope it helps.
see also: https://github.com/mikakaraila/node-red-contrib-opcua/blob/master/102-opcuaclient.js

```
function read(item){
					var ns = item.browseName.namespaceIndex;
					var i = item.nodeId.value;
					
					console.log("namespaceIndex ="+ns+""+"Identifier="+i+"");
					
}

node.session.browse( "ns=0;i=84" ,function (err, itemResults,diagnostics) {
if (err) {
    	      console.log(err);
        	   console.log(itemResults);
            console.log(diagnostics);
          }else{
			for(var i=0; i<itemResults.length; i++){
					itemResults[i].references.forEach(read);
			}
}
```

closed 2016-03-12 by erossignon