node-opcua/node-opcua #98: NodeCrawler after write operation

perok, 2015-07-28:
Hi!

Thanks for accepting and fixing #97. 

Our server implementation is using the Prosys Java SDK and if we try to run the test against our server we will get the same issue as before. I have copied the output from the test below:

````
  NodeCrawler after write
Starting first
first null
read success
starting second
{ identifierType: { [Number: 1] key: 'NUMERIC', value: 1 },
  value: 2294,
  namespace: 0 }
res { value: 0, description: 'No Error', name: 'Good' }
Starting third
    1) should crawl, write to node, and crawl again

  0 passing (10s)
  1 failing

  1) NodeCrawler after write should crawl, write to node, and crawl again:
     Error: timeout of 10000ms exceeded. Ensure the done() callback is being called in this test.
````

As you can see, the first run of the crawler succeeds, but the second run after a write operation will time out. Trying to run the test again will make it fail on the first crawl attempt.

Do you think this is an error on our side with the Prosys Java SDK or a crawler issue?

erossignon, 2015-07-28:
It looks like the server is taking a long time to respond causing the test to timeout ( in your case 10sec)
What is happening if you increase the timeout

perok, 2015-07-29:
I have not tried increasing the timeout. But I have made a [gist](https://gist.github.com/perok/826d64ccad3845190c37) with the output from running ````env DEBUG=node_crawler mocha test -R spec --recursive --grep "NodeCrawler after"````. As you can see the nodecrawler gets alot of data. From line number 1687 you can see that it starts the second run of the nodecrawler after the write operation. The data is there, but it never calls the callback for the read operaton with the result.

perok, 2015-07-29:
I have [diffed](https://gist.github.com/perok/aabca5cbd44479194357) the node_crawler debug output.

erossignon, 2015-07-29:
I wonder if we have found a issue with the prosys demo server.

erossignon, 2016-03-12:
@perok,
I have run  test_e2e_client_node_crawler_crawl_write_crawl against Prosys OPCAU demo Server version 2.2.0.94 and got it working smoothly. 
I'll close the issue for now, feel free to reopen if the problem is still present.

closed 2016-03-12 by erossignon