node-opcua/node-opcua #99: Event "err" Not Working for ClientMonitoredItem

rodp82, 2015-07-31:
Hi,

In the code below, the "changed" event on the monitoredItem works fine, however the "err" event is never triggered when an unknown node ID is supplied.

```javascript
monitoredItem.on('err ', function(statusCode ){
    console.log('an error has occured');
    console.log(statusCode );
});

monitoredItem.on('changed', function(dataValue){
    console.log(dataValue.value.value,);
});
```

The library itself logs a "monitoredItemResult statusCode =  BadNodeIdUnknown (0x80340000)" message. And in the code in lib/client/client_monitored_item.js (line142), the following code is called.
```
self.emit("err", monitoredItemResult.statusCode);
```

Is there something else that I'm missing?

erossignon, 2015-07-31:
You have a trailing space in 'err ' ,it should be 'err'

rodp82, 2015-08-01:
Well spotted. I'd been staring at that for a long time.
Thanks

closed 2015-08-05 by erossignon