node-opcua/node-opcua #100: server.engine.addFolder();

kaltental89, 2015-08-05:
Hi everyone, I'm trying to get the server tutorial (http://node-opcua.github.io/create_a_server.html) running but keep getting following error:

server.engine.addFolder("RootFolder",{ browseName: "MyDevice"}); 
TypeError: undefined is not a function

Any ideas?

kaltental89, 2015-08-05:
Found the problem it should be:
server.engine.createFolder("RootFolder",{ browseName: "MyDevice"}); 

instead of:
server.engine.addFolder("RootFolder",{ browseName: "MyDevice"}); 

Maybe this should be changed in the tutorials.

erossignon, 2015-08-19:
In fact the website has been recently updated to reflect recent changes in 0.48  that were not yet published by the time you found the problem.
version 0.48 has a few breaking changes with earlier version.
please update your node-opcua package version to the latest or use the master version.

closed 2015-08-19 by erossignon