node-opcua/node-opcua #101: problem to browse subfolders in callback function?

zombinary, 2015-08-17:
I have a problem with my recursive callback function. I cannot serialize the asynchronous calls and the browse function reads only one sub-folder. 

```	
var browse = function(item,indx,array,callback){		// differents???
//		console.log("child:\n "+JSON.stringify(callback)+"\n");
//		console.log("callback="+JSON.stringify(callback));
				if(item.isForward == false){
					return 0;		
				}		
				//console.log('|-\ '+'item:\ '+item.nodeId.toString()+',\ name:\ '+item.browseName.name+',\ nodeclass:\ '+item.$nodeClass.key);
				if (item.$nodeClass.key == 'Variable') {
					callback(item);
				}else if (item.$nodeClass.key == 'Object' ) {
						var child = 'ns='+item.browseName.namespaceIndex+';i='+item.nodeId.value;
						opcuaServer_browse(child,function(result){
										callback(result);
						});
				}
	 	}

		var opcuaServer_browse = function(item,callback){
			var buf = [];
			node_browse.session.browse( item,function (err, itemResults,diagnostics) {
     			if (err) {
     	     		console.log(err);
          		console.log(itemResults);
	         	console.log(diagnostics);
         	}else{
         		for(var i=0; i<itemResults.length; i++){
					 	//itemResults[i].references.forEach(browse);
					 	itemResults[i].references.forEach(function(element,index,arr){
						 	//browse(element, index, arr, buf);
						 		browse(element, index, arr, function(item){
							callback(item);
						 	});
					 	});
				    }
	       }
      });
	 } 
	 	      
      
    function OpcUa() {

		var item = "ns=0;i=84";			// '/root'
		
		if (node_browse.client==undefined) {
            node_browse.client = new opcua.OPCUAClient();
            node_browse.subscriptions=subscriptions;
            node_browse.items=items;
        }
        		var buf = [];
        		var i = 0;
				
	  async.series({
		      // First connect to server´s endpoint
            connect: 
							....            
            // Browse
            elements: 
            		function(callback) {
            				opcuaServer_browse(item, function(result){
										callback(null,result);
								});
								//callback();
								//if(opcuaServer_browse(item,buffer) == 1){callback()};
    						},
 			   			// closing session
		     close_session: 
							.....
   ```

erossignon, 2015-08-20:
inside the ```for``` loop in ````opcuaServer_browse ```` the callback is called n times, which is probably not what you want.
you probably need to use  ```async.each```.

zombinary, 2015-08-24:
thanks for your help.
I change the source and now i calling the browse function with the ~~async.each()~~.
But now i get an error in the ~~async.series~~, if i call there ~~console.log(elements.result);~~,  i get the message ~~TypeError: Converting circular structure to JSON~~.
How can i wait for the end of the process?
Must i create an child process?

```
var browse = function(item,indx){
				var child = 'ns='+item.browseName.namespaceIndex+';i='+item.nodeId.value;
		
				if (item.$nodeClass.key == 'Variable') {
					indx({"Variable" : item.browseName.name});
					
				}else if (item.$nodeClass.key == 'Object' ) {
					if(item.isForward){
						
						//indx({"Object" : item.browseName.name});
						
/*						opcuaServer_browse(child,function(result){
									buf.push(result);
						});*/
						
						async.each(child, function(elementChild, callback){
								opcuaServer_browse(child,function(result){
												sub.push(result);
												callback();							
								});
							},
							function(err){
						});
						indx(buf.push(sub));
			}
	 	}
}	 	

		var opcuaServer_browse = function(item,callback){
			node_browse.session.browse( item,function (err, itemResults,diagnostics) {
     			if (err) {
     	     		console.log(err);
          		console.log(itemResults);
	         	console.log(diagnostics);
         	}else{
						async.each(itemResults[0].references,function( elementChild,callback){
										browse(elementChild,function(element){
										buf.push(element);
										callback();	
								});
							},
							function(err){
		                			console.log(new Error(err));
						}); 
	       }
	       callback(buf);
      });
	 } 
```

erossignon, 2015-08-24:
Remember that the callback function must be called exactly once inside your async function (here opcuaServer_browse) regardless of the scenario. 

The typical callback function takes a Error object as a first argument to signal  that the async function has failed or null if it succeed and in this case a additional ```data``` object is passed as a second argument. The callback function signature is ```callback(err,data);```

````async.each```
You need to call the callback method inside the final function. ```async.each(data,itemFunc,finalFunc)```, This will make sure that the callback is called only once the iteration is completed.

```javacript 
var opcuaServer_browse = function (item, callback) {
    node_browse.session.browse(item, function (err, itemResults, diagnostics) {
        if (err) {
            console.log(err);
            console.log(itemResults);
            console.log(diagnostics);
            callback(err);
        } else {
            async.each(itemResults[0].references, function (elementChild, inner_callback) {
                browse(elementChild, function (element) {
                    buf.push(element);
                    inner_callback();
                });
            }, function (err) {
                callback(err,buf);
            });
        }
    });
} 
```

zombinary, 2015-08-26:
Thanks, with async.each it works.
I get the right printout from the Parentfolder ('result = ...') and after there the correct printout from the variable in the subfolders. I think i have a faulty reasoning but it's solvable.
Here is my code maybe you have a idee to solve it.

'''
var browse = function (item, indx){
	 //console.log("item:\ "+item);
	 var sub = [];	
	 var child = 'ns=' + item.browseName.namespaceIndex + ';i=' + item.nodeId.value;
		
	 if (item.$nodeClass.key == 'Variable') {
        indx({"Variable": item.browseName.name});			//callback
    } else if (item.$nodeClass.key == 'Object' || item.$nodeClass.key == 'Method') {
        if (item.isForward) {

      node_browse.session.browse( child,function (err, itemResults,diagnostics) {
      		var sub = [];
      		var i = true;
      		var parentname = "unknown";
      		
      		if (err) {
     	     		console.log("ERROR child:\ "+err);
          		console.log(itemResults);
	         	console.log(diagnostics);
	         	callback(err);
         	}else{
				   async.each(itemResults[0].references, function(child, child_inner_callback){
							// ignore folder 'Views'
							if(child != "ns=0;i=87"){
								browse(child, function(child_element){
									sub.push(child_element); 							//result= [{"Object":[{"Variable":"ServerArray"},{"Variable":"NamespaceArray"},{"Variable":"ServerStatus"},{"Variable":"ServiceLevel"},{"Variable":"Auditing"}]}]			
								});
							}
							child_inner_callback();
						},function(err) {
			  				console.log('subbuf:\n'+ JSON.stringify(sub));
			  				indx({"Object": sub});
				  });
				}
			 });

			}
		 }
		};	

 var opcuaServer_browse = function(item,callback){
		//console.log("item=\ "+item);
			var buf = [];
		
			node_browse.session.browse( item,function (err, itemResults,diagnostics) {
     			if (err) {
     	     		console.log(err);
          		console.log(itemResults);
	         	console.log(diagnostics);
	         	callback(err);
         	}else{
//         		 	itemResults[0].references.forEach(function(element,index,arr){
//						 	browse(element, index, arr,subfolder);
					async.each(itemResults[0].references, function(item, inner_callback){
									//console.log("parent:\ "+item.browseName.name);
									//buf.push(item.browseName.name);
									
									//opcuaServer_browse(item, function(element){
									browse(item, function(element){
										console.log("parent:\n"+JSON.stringify(element));
										buf.push(element);
									});
									inner_callback();
									},function(err) {
			  								console.log('buf:\n'+ JSON.stringify(buf));
			  								callback(err, buf);
						});
				}
		});	       
	};
'''

zombinary, 2015-08-27:
I solved the problem.
It was a problem with the callback in ```async.series``` function,
with ```setTimeout()``` i freezed these. I think it is dirty but it works.

```
				// Browse
				elements: 
            			function(callback) {
            				var buf = [];
            				var rslt = []; 
								opcuaServer_browse(item,buf,function(err,result){
										setTimeout(function(){
										callback(err,result);},5000);
								});
							},
```

closed 2015-08-28 by zombinary