node-opcua/node-opcua #103: AssertionError: parentObject instanceof BaseNode

rodp82, 2015-08-17:
Hi everyone, 

I'm working through the server tutorial but keep getting the following error message when trying to run the script:

```
AssertionError: parentObject instanceof BaseNode
    at AddressSpace.addVariable (/Projects/OPCServerSimulator/node_modules/node-opcua/lib/address_space/address_space.js:690:5)
    at ServerEngine.addVariable (/Projects/OPCServerSimulator/node_modules/node-opcua/lib/server/server_engine.js:876:31)
    at construct_my_address_space (/Projects/OPCServerSimulator/index.js:31:46)
    at post_initialize (/Projects/OPCServerSimulator/index.js:83:5)
    at Object._onImmediate (/Projects/OPCServerSimulator/node_modules/node-opcua/lib/server/opcua_server.s:413:9)
    at processImmediate [as _immediateCallback] (timers.js:363:15)
```

Any ideas?

erossignon, 2015-08-19:
Hi, tutorials was updated before v0.0.48 was published. Please  make sure to  install the latest node-opcua version (>=0.0.48) and let me know.

rodp82, 2015-08-19:
Yeah, that sorted the issue.
Thanks very much

closed 2015-08-19 by rodp82