node-opcua/node-opcua #104: AddressSpace uses NodeIds twice

janschatz, 2015-08-17:
The method `AddressSpace._build_new_NodeId()` generates NodeIds in the namespace with index 1, because `AddressSpace._private_namespace` is 1. If I have loaded a nodeset XML file which also defines NodeIds in the namespace with index 1, it happens that `AddressSpace._build_new_NodeId()` generates a NodeId that is already being used by a node from the XML file. How should I resolve this? Is it OK to change `AddressSpace._private_namespace` to any other value?

janschatz, 2015-08-17:
Mybe `AddressSpace._build_new_NodeId()` should be modified like this?
```javascript
var nodeId;
do {
    nodeId = makeNodeId(this._internal_id_counter, this._private_namespace);
    this._internal_id_counter += 1;
} while(this._nodeid_index.hasOwnProperty(nodeId));
return nodeId;
```

janschatz, 2015-09-07:
@erossignon, if it is OK for you, I will commit the above fix.

erossignon, 2015-09-07:
Yes, feel free to submit a pull request for code + associated test

closed 2015-09-09 by erossignon