node-opcua/node-opcua #105: UAObjectType.instantiate() does not accept subtypes of FolderType in options.organizedBy

janschatz, 2015-08-18:
I have subtyped `FolderType`. Let's assume the subtype is called `MyFolderType`. Now I want to instantiate another object custom type `MyOtherType` and let the object be organised by an object of `MyFolderType`.

```javascript
MyOtherType.instantiate({ organizedBy: 'MyFolderTypeInstance', browseName: 'item' });
```

An assertion in `AddressSpace.getFolder()` fails that checks whether `MyFolderTypeInstance` is of the `FolderType` type.

```javascript
folderTypeId = self._coerceTypeDefinition("FolderType");
assert(folderTypeId, " ????");
assert(folder.hasTypeDefinition.toString() === folderTypeId.toString(), "expecting a Folder here ");
```

Shouldn't the method also allow subtypes of `FolderType`?

erossignon, 2015-08-19:
Yes it should ! Current implementation only cope with FolderType type of folders.

erossignon, 2015-09-05:
In fact I think there is a confusion here: what do you think of this little test. Is that what you have in mind ?

```javascript 
    it("should be possible to create an object organized by a folder whose type is a subtype of FolderType",function(){

        var temperatureSensorType = createTemperatureSensorType(address_space);

        //xx var folderType = address_space.findObjectType("FolderType");

        var myFolderType = address_space.addObjectType({browseName: "MyFolderType",subtypeOf: "FolderType"});

        myFolderType.browseName.toString().should.eql("MyFolderType");
        myFolderType.subtypeOfObj.browseName.toString().should.eql("FolderType");

        // now create a folder of type MyFolderType inside the RootFolder
        var myFolder = myFolderType.instantiate({browseName:"MyFolder",organizedBy:"RootFolder" });

        // now create a simple var inside the new folder (method 1)
        var myObject = address_space.addVariable(myFolder,{browseName:"Obj1",dataType:"Double"});
        myObject.browseName.toString().should.eql("Obj1");

        // now create a simple var isnide the new folder (method 2)
        var myObject2 = temperatureSensorType.instantiate({ browseName:"Obj2", organizedBy: myFolder});

        myObject2.browseName.toString().should.eql("Obj2");

    });
```

closed 2015-09-07 by janschatz

janschatz, 2015-09-07:
Yes, that was what I intended to do. Thank you very much!