node-opcua/node-opcua #106: Raspberry Pi 2 - HDA capability

jcodling, 2015-08-18:
Greetings everyone,

I have the node OPC UA server running on a Raspberry Pi 2 exposing two simulated variables and an ultrasonic sensor. It works great using a modified node OPC UA simple-client with a RESTful API to a custom web front end displaying the results. I am currently involved in a project to have an OPC UA server running in low cost packages expose remote sensor data. I require HDA functionality built into the server in case the connection is severed so that data is not lost while the remote package is still functional and the connection is re-established. Once reconnected the server should return the missing data and continue with the subscription.

I have looked at the code and documentation for the Historian capability. Is there anyone who has implemented HDA functionality using node OPC UA with this kind of capability or can point me in a direction to implement this feature?

This node OPC UA project gave me a great headstart. Thank you.

damienclaveau, 2015-10-26:
Hi Jeff,

I'm looking for the same feature.
Are you going to work on it in your forker repository ?

Damien

jcodling, 2015-10-26:
I tried but didn't have any luck figuring out the current code enough to implement it. I've moved on to using the Open OPC-UA package which can be compiled for ARM7 and implements all the standards.

erossignon, 2018-01-25:
HDA is now implemented as in memory storage and can be extended to stored to custom database oif necessary.

closed 2018-01-25 by erossignon