node-opcua/node-opcua #107: Optional properties of custom ObjectTypes

janschatz, 2015-08-19:
What is the correct way to deal with optional properties of custom ObjectTypes?
I instance the ObjectType like this:
```javascript
var MyType = address_space.findObjectType('MyType');
var myObject = MyType.instantiate({ organizedBy: parentNode, browseName: 'name' });
```
Then I check if I need certain optional properties and try to delete non-needed ones like this:
```javascript
var prop = myObject.getPropertyByName('prop');
address_space.deleteObject(prop.nodeId);
```

The function `deleteObject()` does not sound like deleting properties, but the source code looks like it is able to deal with several NodeClasses. So I gave it a try. However it does not remove HasProperty references from `myObject`. Should I remove these manually?

erossignon, 2015-08-19:
We probably need to create a specific APi for this , something like:

```javascript
var prop = myObject.getPropertyByName('prop');
myObject.removeProperty(prop.nodeId);
```

janschatz, 2015-08-20:
How do you like the following implementation?
```javascript
/**
 * removes a property, i.e. the property node and the HasProperty reference
 * @method removeProperty
 * @param  propertyNodeId
 *         The node ID of the property node
 */
BaseNode.prototype.removeProperty = function(propertyNodeId) {
    // Search for HasProperty reference with given target node ID
    for(var i = 0; i < this._references.length; ++i) {
        if((this._references[i].referenceType === "HasProperty") &&
           this._references[i].isForward &&
           (this._references[i].nodeId === propertyNodeId)) {
            // Remove the reference
            this._references.splice(i, 1);
            // Decrement loop counter because we altered the array length
            --i;
        }
    }

    // Delete property node
    this.__address_space.deleteObject(propertyNodeId);
};
```

janschatz, 2015-09-07:
@erossignon, if you don't object, I will commit the above implementation.

erossignon, 2015-09-07:
Yes , fill free to provide a pull request for it, including  code + associated unit test.

janschatz, 2015-09-10:
While implementing the unit test for the new function I found that it doesn't really solve the problem. In fact ```AddressSpace.deleteObject()``` already removes the references from the parent node through ```object.unpropagate_back_references(self);```.

The problem is the ```_properties``` cache of the parent object. This cache is build once when ```BaseNode.getProperties()``` is called, but it is never updated.
I think this redundant storage is a general problem.

Let me illustrate this by extending [this](https://github.com/node-opcua/node-opcua/blob/master/test/address_space/test_variable.js#L149-L159) test case:
```javascript
    it("addProperty should accept a typeDefinition as a String", function () {
        // rootFolder.getProperties();

        var nodeVar = the_address_space.addProperty(rootFolder, {
            browseName: "SomeVariable5",
            dataType: "Double",
            typeDefinition: "PropertyType"
        });
        nodeVar.hasTypeDefinition.should.be.instanceOf(NodeId);
        nodeVar.hasTypeDefinition.toString().should.eql("ns=0;i=68");

        nodeVar = rootFolder.getPropertyByName("SomeVariable5");
        nodeVar.should.be.instanceOf(UAVariable);
    });
```
If you uncomment ```rootFolder.getProperties();```, this will build ```rootFolder```'s ```_properties``` cache and it won't be updated when the new property is added. So ```rootFolder.getPropertyByName("SomeVariable5");``` will fail to find the new property.

erossignon, 2015-09-10:
Caches are used to speedup frequent operations for which results is unlikely to change and for which the cost of data extraction is relatively high.
We need to change the code so that adding a property or deleting a property shall cause the object to clear its cache , so that it can be reconstructed when needed.
For instance :
```javascript
BaseNode.prototype._add_backward_reference = function (reference) {
    var self = this;
    [...]
    self._clearInternalCaches();     //<< this line added
    [...]
    if (find_reference(self._references, reference).length > 0) {
        return;
    }
   
    self._back_references.push(reference);

};
BaseNode.prototype._clearInternalCaches = function() {
    self._properties = null;
}; 
```

erossignon, 2015-12-02:
Now ObjectType#instantiate and VariableType#instantiate only create properties marked with ModellingRule="Mandatory". Properties with ModellingRule="Optional" can be requested this way:

```
       var channel2 = analyserChannelType.instantiate({
            browseName: "__Channel2",
            optionals: [ "ParameterSet" ]
        });
 ```
as shown in this [test](D:\projet\node-opcua\test\end_to_end\companions\analyser_device_integration\test_e2e_server_with_analyser_device.js)

closed 2015-12-02 by erossignon