node-opcua/node-opcua #108: UAObjectType.instantiate incomplete for complex ObjectTypes like DI and ADI

eoursel, 2015-08-19:
Can't complete the instantiation of a FTNIR ADI Device. 
```javascript
   var ftnirType = server.engine.address_space.findObjectType("4:FTNIRSimulatorDeviceType"); 
    var ftnirInstance = ftnirType.instantiate({ browseName: "MyFTNIR", organizedBy: deviceSet });
```

```
D:\Users\Eric\Git\Github\node-opcua\lib\address_space\base_node.js:1098
        var clone = node.clone();
                         ^
TypeError: undefined is not a function
    at D:\Users\Eric\Git\Github\node-opcua\lib\address_space\base_node.js:1098:26
    at Array.forEach (native)
    at UAObjectType.BaseNode._clone_references (D:\Users\Eric\Git\Github\node-opcua\lib\address_space\base_node.js:1097:16)
    at initialize_properties_and_components (D:\Users\Eric\Git\Github\node-opcua\lib\address_space\ua_object_type.js:112:22)
    at initialize_properties_and_components (D:\Users\Eric\Git\Github\node-opcua\lib\address_space\ua_object_type.js:109:9)
    at initialize_properties_and_components (D:\Users\Eric\Git\Github\node-opcua\lib\address_space\ua_object_type.js:109:9)
    at initialize_properties_and_components (D:\Users\Eric\Git\Github\node-opcua\lib\address_space\ua_object_type.js:109:9)
    at UAObjectType.instantiate (D:\Users\Eric\Git\Github\node-opcua\lib\address_space\ua_object_type.js:119:5)
    at OPCUAServer.<anonymous> (D:\Users\Eric\Git\Github\node-opcua\bin\di_server.js:101:35)
    at OPCUAServer.emit (events.js:104:17)
    at Immediate._onImmediate (D:\Users\Eric\Git\Github\node-opcua\lib\server\opcua_server.js:427:14)
    at processImmediate [as _immediateCallback] (timers.js:358:17)
PS D:\Users\Eric\Git\Github\node-opcua>
```

Looks like the this object type contains a ConnectionPointType according to the DI definition and the current code can't handle this type of deep structure.

erossignon, 2015-08-24:
Can you create a minimalist unit test to highlight the issue ?

erossignon, 2015-10-18:
It looks like the ```node-opcua\nodesets\Opc.Ua.Di.NodeSet2.xml``` was out of of sync and invalid.

closed 2016-03-23 by erossignon