node-opcua/node-opcua #109: Assertions in crypto_utils.js

janschatz, 2015-08-20:
I just tried connecting as a non-anonymous user for the first time. Two assertions failed in `lib/misc/crypto_utils.js`.
The first one is in `makeMessageChunkSignature()`.
```javascript
assert(identifyPemType(options.privateKey) === "RSA PRIVATE KEY");
```
The PEM type of my private key is "PRIVATE KEY". I generated it like described in the server tutorial using OpenSSL 1.0.1p (2015-07-09) for Win64.

After commenting this assertion another one failed in `makeMessageChunkSignatureWithDerivedKeys()`.
```javascript
assert(signature.length = derivedKeys.signatureLength);
```
I think this should not be an assignment but rather a comparison.

erossignon, 2015-08-31:
http://stackoverflow.com/questions/20065304/what-is-the-differences-between-begin-rsa-private-key-and-begin-private-key

janschatz, 2015-09-04:
This explains the difference. But I still don't understand why the same commands produce different key files for me. Anyway, it seems to work with both types of key files. So I think you could allow "PRIVATE KEY", too.

closed 2015-09-07 by janschatz