node-opcua/node-opcua #110: Problems concerning secure channel ID and keep alive timeout

janschatz, 2015-08-20:
I constantly see the follwoing messages on the log of my OPC UA server.
```
ERROR: channel.secureChannelId !== session.secureChannelId 3 2
ERROR: channel.secureChannelId !== session.secureChannelId 4 3
ERROR: channel.secureChannelId !== session.secureChannelId 5 4
ERROR: channel.secureChannelId !== session.secureChannelId 6 5
ERROR: channel.secureChannelId !== session.secureChannelId 7 6
ERROR: channel.secureChannelId !== session.secureChannelId 8 7
closing SESSION  urn:cde142:UnifiedAutomation:UaExpert@cde142  because of timeout =  30000  has expired without a keep alive
closing SESSION  urn:cde142:UnifiedAutomation:UaExpert@cde142  because of timeout =  30000  has expired without a keep alive
ERROR: channel.secureChannelId !== session.secureChannelId 9 8
closing SESSION  urn:cde142:UnifiedAutomation:UaExpert@cde142  because of timeout =  30000  has expired without a keep alive
ERROR: channel.secureChannelId !== session.secureChannelId 10 9
```
What does it mean? How can I deal with it?
I am monitoring a variable that changes every second. So I don't understand why the server is expecting any keep alive messages.

Thanks for your support so far!

janschatz, 2015-08-21:
Could you reproduce the behaviour?

erossignon, 2015-08-24:
yes, I can. We need further investigation.

janschatz, 2015-09-04:
Now it happens far less often, but I still see something like the following once in a while:
```
Session is being transferred from channel 4  to channel  5
ERROR: channel.secureChannelId !== session.secureChannelId 5 4
```

I also see the following when the client reconnects:
```
EXCEPTION CAUGHT WHILE PROCESSING REQUEST !! ActivateSessionRequest
 display_trace_from_this_projet_only =
value cannot be coerced to DataType: UtcTime

 messages  EXCEPTION CAUGHT WHILE PROCESSING REQUEST !!! ActivateSessionRequest
value cannot be coerced to DataType: UtcTime
Error: value cannot be coerced to DataType: UtcTime
    at Variant._enum_properties.dataType.set [as dataType] (<project path>\node_modules\node-opcua\_generated_\_auto_generated_Variant.js:41:29)
    at new Variant (<project path>\node_modules\node-opcua\_generated_\_auto_generated_Variant.js:108:19)
    at <project path>\node_modules\node-opcua\lib\address_space\address_space_add_event_type.js:160:46
    at Array.forEach (native)
    at initialize_properties_and_components (<project path>\node_modules\node-opcua\lib\address_space\address_space_add_event_type.js:153:27)
    at initialize_properties_and_components (<project path>\node_modules\node-opcua\lib\address_space\address_space_add_event_type.js:148:13)
    at initialize_properties_and_components (<project path>\node_modules\node-opcua\lib\address_space\address_space_add_event_type.js:148:13)
    at AddressSpace.constructEventData (<project path>\node_modules\node-opcua\lib\address_space\address_space_add_event_type.js:172:9)
    at UAObject.raiseEvent (<project path>\node_modules\node-opcua\lib\address_space\ua_object_raiseEvent.js:28:39)
    at OPCUAServer.raiseEvent (<project path>\node_modules\node-opcua\lib\server\opcua_server.js:1038:16)
```
I don't know if this is related to each other.

janschatz, 2015-09-07:
Thanks, the message concerning UtcTime has not appeared any more since your fix.

closed 2015-10-01 by erossignon