node-opcua/node-opcua #111: Beginner, startup issue

Laro88, 2015-08-21:
I wanted to give node-opcua a run as a client, but I hit an initial crash when trying to connect to a existing server.
The server I have is explorable using UaExpert 1.2.2.175
Is there any guidance on getting / testing the connection before any interaction is started, I am suspecting there might be some security hickup although the opc server should be running without any security.
[image]

janschatz, 2015-08-21:
The yellow text is just some hints concerning performance.
Is app.js your application? What is happening in line 48? Seems like you are accessing an invalid object there.

Laro88, 2015-08-21:
The app is the client example with the ip of a running server.
The fault seems to be from behind the scenes what I suppose is in async connection establishment. debugging is behaving a bit weird so I have a hard time determining the actual origin.
Is there some examples on connection establishment with regards to authentication? My server is set to near zero, however I might have overlooked something.

Laro88, 2015-08-24:
The demo app is working fine, VS2013 has some sort of problem debugging the opcua project, so my read address typo resulted in the error appearing elsewhere.
```javascript
//this works :-)
var _var = "ns=2;s=EXX310_00035.Raw_sensor.Sensor 2.Value";
        the_session.readVariableValue(_var, function (err, dataValues, diagnostics) {
            if (err)
                logger.error(_var + err);
            else {
                logger.info(_var , dataValues[0].value.value);
            }
            callback(err);
        })
```

closed 2015-08-24 by Laro88