node-opcua/node-opcua #112: Problems with ByteString values in io.js

janschatz, 2015-08-21:
I changed from node.js 0.10.4 to io.js 3.1.0 some days ago, because I had problems with some cryptographic functions.
Now I noticed that values of data type ByteString cannot be read any more. I get the following output on the console:
```
EXCEPTION CAUGHT WHILE PROCESSING REQUEST !! ReadRequest
 display_trace_from_this_projet_only =
buf instanceof Buffer

 messages  EXCEPTION CAUGHT WHILE PROCESSING REQUEST !!! ReadRequest
buf instanceof Buffer
AssertionError: buf instanceof Buffer
    at BinaryStream.writeByteStream (<project path>\node_modules\node-opcua\lib\misc\binaryStream.js:311:5)
    at exports.encodeByteString (<project path>\node_modules\node-opcua\lib\misc\encode_decode.js:623:12)
    at Object.Variant_Schema.encode (<project path>\node_modules\node-opcua\schemas\Variant_schema.js:232:7)
    at Variant.encode (<project path>\node_modules\node-opcua\_generated_\_auto_generated_Variant.js:143:11)
    at Object.DataValue_Schema.encode (<project path>\node_modules\node-opcua\schemas\DataValue_schema.js:70:29)
    at DataValue.encode (<project path>\node_modules\node-opcua\_generated_\_auto_generated_DataValue.js:115:11)
    at <project path>\node_modules\node-opcua\_generated_\_auto_generated_ReadResponse.js:98:63
    at exports.encodeArray (<project path>\node_modules\node-opcua\lib\misc\encode_decode.js:785:9)
    at ReadResponse.encode (<project path>\node_modules\node-opcua\_generated_\_auto_generated_ReadResponse.js:98:5)
    at MessageChunker.chunkSecureMessage (<project path>\node_modules\node-opcua\lib\misc\message_chunker.js:65:13)
```
The getter function returns a `Buffer`. But somewhere it is converted to a `Uint8Array`. I cannot remember changing anything that could cause this porblem. So I think it is io.js.

erossignon, 2015-08-24:
I stumbled across something similar with iojs 3.0. I had to fix it this way (see  942ded50965a511a1c8fae26d93174469a8ebb24 ) to make the tests pass. It looks like there is more work to do here to cope with recent breaking changes introduced in iojs.

erossignon, 2015-08-24:
@janschatz can you confirm which version of node-opcua you've been using when you found this issue ? was the fix 942ded50965a511a1c8fae26d93174469a8ebb24 already in ?
If yes, I will need more detail to help me reproduce the issue.

janschatz, 2015-09-04:
Seems like this fix was not present. I pulled the latest version and now it passes the assertion. Thank you!

closed 2015-09-04 by janschatz