node-opcua/node-opcua #113: Default access levels

janschatz, 2015-08-21:
I have created a nodeset.xml file with the UaModeler by Unified Automation. The tool seems to assume that the default access level is `CurrentRead`, because it does not include any access level specification for nodes that have this combination.
However node-opcua seems to assume that `CurrentRead | CurrentWrite` is the default access level. The result is that all nodes that should be read-only are read-write in my server.
I guess the UaModeler is right, seeing that the node-opcua server also allows write-access to the server's build information etc (see `Opc.Ua.NodeSet2.xml`).

erossignon, 2015-08-24:
`Well spotted. AccessLevel and UserAccessLevel attributes are not accounted for in ``lib\address_space\load_nodeset2.js```` at this time. This shall be fixed.

janschatz, 2015-09-04:
Now access levels are correctly parsed - not only from nodeset2.xml, but also from custom nodeset files.

closed 2015-09-04 by janschatz