node-opcua/node-opcua #114: browseName and displayName are missing when nodeId is numeric

holymoly, 2015-08-27:
Hi,

Thanks for this awesome package. 
I have a question regarding the session.browse function.
I modified the sample_client.js for playing around with browsing.
When I use this code the result does not have the browseName and displayName as a String.

```
var browseDescription = {
          nodeId: 84,
          referenceTypeId: 0,
          browseDirection: 2
       }
       the_session.browse(browseDescription, function(err,browse_result,diagnostics){
           if(!err) {
               //console.log( browse_result);
               browse_result[0].references.forEach(function(reference) {
                   console.log( "------------------");
                   console.log( reference)
                   //console.log( reference.browseName.namespaceIndex);
                   //console.log( reference.nodeId.value);
                   console.log( "------------------");
               });
           }else{
            console.log(err);
            console.log(browse_result.toString());
            console.log(diagnostics);
           }
           callback(err);
       });
```
Result:
```
{ referenceTypeId: 
   { identifierType: { [Number: 1] key: 'NUMERIC', value: 1 },
     value: 0,
     namespace: 0 },
  isForward: false,
  nodeId: 
   { identifierType: { [Number: 1] key: 'NUMERIC', value: 1 },
     value: 61,
     namespace: 0,
     namespaceUri: null,
     serverIndex: 0 },
  browseName: { namespaceIndex: 0, name: null },
  displayName: { text: null, locale: null },
  '$nodeClass': { [Number: 0] key: 'Unspecified', value: 0 },
  typeDefinition: 
   { identifierType: { [Number: 1] key: 'NUMERIC', value: 1 },
     value: 0,
     namespace: 0,
     namespaceUri: null,
     serverIndex: 0 } }
```
When is use "RootFolder" i get the Strings.
```
the_session.browse("RootFolder", function(err,browse_result,diagnostics){
           if(!err) {
               //console.log( browse_result);
               browse_result[0].references.forEach(function(reference) {
                   console.log( "------------------");
                   console.log( reference)
                   //console.log( reference.browseName.namespaceIndex);
                   //console.log( reference.nodeId.value);
                   console.log( "------------------");
               });
           }else{
            console.log(err);
            console.log(browse_result.toString());
            console.log(diagnostics);
           }
           callback(err);
       });
```
Result:
```
{ referenceTypeId: 
   { identifierType: { [Number: 1] key: 'NUMERIC', value: 1 },
     value: 35,
     namespace: 0 },
  isForward: true,
  nodeId: 
   { identifierType: { [Number: 1] key: 'NUMERIC', value: 1 },
     value: 85,
     namespace: 0,
     namespaceUri: null,
     serverIndex: 0 },
  browseName: { namespaceIndex: 0, name: 'Objects' },
  displayName: { text: 'Objects', locale: null },
  '$nodeClass': { [Number: 1] key: 'Object', value: 1 },
  typeDefinition: 
   { identifierType: { [Number: 1] key: 'NUMERIC', value: 1 },
     value: 61,
     namespace: 0,
     namespaceUri: null,
     serverIndex: 0 } }
```
Is this a normal behavior? I just started playing with this one:

opc.tcp://commsvr.com:51234/UA/CAS_UA_Server

erossignon, 2015-08-27:
you need to specify resultMask in browseDescription 
for instance 

```javascript
var browseDescription = {
            nodeId: "ns=0;i=84",
            browseDirection: BrowseDirection.Inverse,
            nodeClassMask: 0, // 0 = all nodes
            referenceTypeId: "Organizes",
            resultMask: 0x3F
        };
```
see https://github.com/node-opcua/node-opcua/blob/master/test/end_to_end/test_opcua_ClientServer_UAAnalogItem.js#L119 for instance

erossignon, 2015-08-27:
see also http://node-opcua.github.io/api_doc/classes/BrowseDescription.html

closed 2015-08-27 by holymoly

holymoly, 2015-08-27:
resultMask:
https://github.com/node-opcua/node-opcua/blob/95886770c78dfb6ecd408530b3d3426b18ac85ea/schemas/ResultMask_enum.js

erossignon, 2015-08-31:
you can also create a resultMask using the helper function
var resultMask = opcua.browse_service.makeResultMask("ReferenceType | IsForward | BrowseName | NodeClass | TypeDefinition");