node-opcua/node-opcua #116: subribe to item to receive value change

bueffeljaeger, 2015-09-10:
Hi,

I want to receive changes of the value of an item by using subscriptions. For testing I'm using Matrikons [OPC Server for Simulation](https://www.matrikonopc.com/products/opc-drivers/opc-simulation-server.aspx) in combination with [UA Wrapper for COM OPC Servers](https://www.matrikonopc.com/opc-ua/products/ua-wrapper-com-opc-servers.aspx) to use opc-ua as a client. To verify the connection and the value I want to access, I use the [UaExpert](https://www.unified-automation.com/downloads/opc-ua-clients.html) client from unified automation (running on the same linux machine where the node client is running).

This is the modified code:
```javascript
var opcua = require("node-opcua");
var async = require("async");

var client = new opcua.OPCUAClient({requestedSessionTimeout: 20000});

var endpointUrl = "opc.tcp://192.168.76.130:21381/MatrikonOpcUaWrapper";
var the_session, the_subscription;

async.series([

        // step 1 : connect to
        function(callback)  {
            client.connect(endpointUrl,function (err) {
                if(err) {
                    console.log(" cannot connect to endpoint :" , endpointUrl );
                } else {
                    console.log("connected !");
                }
                callback(err);
            });
        },

        // step 2 : createSession
        function(callback) {
            client.createSession( function(err,session) {
                if(!err) {
                    the_session = session;
                }
                callback(err);
            });
        },
        // step 4 : read a variable with readVariableValue
        function(callback) {
            the_session.readVariableValue("ns=2;s=i.Test", function(err,dataValues) {
                if (!err) {
                    console.log(" % Test readVariableValue = " , dataValues);
                }
                callback(err);
            });
        },

        // step 5: install a subscription and install a monitored item for 10 seconds
        function(callback) {
            the_subscription=new opcua.ClientSubscription(the_session,{
                requestedPublishingInterval: 500,
                requestedLifetimeCount: 10,
                requestedMaxKeepAliveCount: 5,
                maxNotificationsPerPublish: 10,
                publishingEnabled: true,
                priority: 1
            });

            the_subscription.on("started",function(){
                console.log("subscription started for 5 seconds - subscriptionId=",the_subscription.subscriptionId);
            }).on("keepalive",function(){
                console.log("keepalive");
            }).on("terminated",function(){
                callback();
            });

            setTimeout(function(){
                the_subscription.terminate();
            },10000);

            // install monitored item
            var monitoredItem  = the_subscription.monitor(
                {
                    nodeId: opcua.resolveNodeId("ns=2;s=i.Test"),
                    attributeId: 7
                },
                {
                    samplingInterval: 250,
                    discardOldest: true,
                    queueSize: 1
                },
                opcua.read_service.TimestampsToReturn.Both
            );

            console.log("-------------------------------------");

            monitoredItem.on("changed",function(dataValue){
                console.log(" % i.Test changed = ",dataValue );
            });
        },
        // close session
        function(callback) {
            the_session.close(function(err){
                if(err) {
                    console.log("session closed failed ?");
                }
                callback();
            });
        }

    ],
    function(err) {
        if (err) {
            console.log(" failure ",err);
        } else {
            console.log("done!");
        }
        client.disconnect(function(){});
    }) ;
```
Connecting, creating the session, browsing and reading the value using readVariableValue worked out of the box but I never receive an update for the subscribed item (which changes every view seconds). I do get a single callback on the "onchange" callback but that's it. And I can't find the value like I get with readVariableValue. Am I doing something wrong is this not the way it works?

Here is the output of the node client:
```
connected !
 % Test readVariableValue =  [ { value: { '$dataType': [Object], '$arrayType': [Object], value: 64 },
    statusCode: { value: 0, description: 'No Error', name: 'Good' },
    sourceTimestamp: null,
    sourcePicoseconds: 0,
    serverTimestamp: null,
    serverPicoseconds: 0 } ]
-------------------------------------
subscription started for 5 seconds - subscriptionId= 3
 % i.Test changed =  { value: 
   { '$dataType': { [Number: 3] key: 'Byte', value: 3 },
     '$arrayType': { [Number: 0] key: 'Scalar', value: 0 },
     value: 1 },
  statusCode: { value: 0, description: 'No Error', name: 'Good' },
  sourceTimestamp: null,
  sourcePicoseconds: 0,
  serverTimestamp: Thu Sep 10 2015 09:37:17 GMT+0200 (CEST),
  serverPicoseconds: 0 }
keepalive
keepalive
keepalive
done!
```
And is it possible to subscribe to a list of items of do I have to do it item by item?

Here are screensots from the OPC UA Server settings:
[image]

And here you can see the Wrapper screen:
[image]

I do get a the correct values, even with subscription, with the uaexpert client running on linux:
[image]

And hint is welcome!

Thanks,
Max

erossignon, 2015-09-12:
²
you probably want to monitor attributeId: 13 (Value) instead of attributeId: 7 ( UserWriteMask)

```javascript
            // install monitored item
            var monitoredItem  = the_subscription.monitor(
                {
                    nodeId: opcua.resolveNodeId("ns=2;s=i.Test"),
                    attributeId: opcua.AttributeIds.Value  // << here
                },
                {
                    samplingInterval: 250,
                    discardOldest: true,
                    queueSize: 1
                },
                opcua.read_service.TimestampsToReturn.Both
            );
```

erossignon, 2015-09-12:
note for a better output you can use .toString() to get rich formatting of your object in the debug console.

``` 
console.log(" % i.Test changed = ",dataValue );
console.log(" % i.Test changed = ",dataValue.toString() ); // better
```

bueffeljaeger, 2015-09-14:
Perfect - it works! Thanks, this saved me a lot of time. I already wondered what this attributeid is for a searched through the coed. Now I know.

Is there an elegant way to subscribe to a lot (about 50) of items?

erossignon, 2015-09-14:
have a look at [How to subscribe to more than one monitored item?](https://github.com/node-opcua/node-opcua/issues/69)

closed 2016-03-12 by erossignon

LoicTryToDev, 2017-10-05:
Hi,
I use node-opcua since a few weeks now. It's really powerfull and exactly what I was looking for.
I still have some issues finding documentations on this product. Maybe you can tell me where I could find a list of functions and properties (with values they can take).

For now, I'd like to know How I can detect that the opcua.ClientSubscription is down ?

            async.series([

            // step 1 : connect to
                function(callback) {
                    console.log(" connecting to ", endpointUrl.cyan.bold);
                    client.connect(endpointUrl,callback);
                },

            // step 2 : créé une session client OPCUA
                function(callback) {

                    client.createSession(userIdentity,function (err,session){
                        if (!err) {
                            the_session = session;
                            console.log(" session created".yellow);
                        }
                        callback(err);
                    });
                },

            // step 5: Abonne la session au serveur OPC UA cible (ici Panorama)
                function(callback) {
                    the_subscription=new opcua.ClientSubscription(the_session,{
                        requestedPublishingInterval: 200,
                        requestedMaxKeepAliveCount:  200,
                        requestedLifetimeCount:      600,
                        maxNotificationsPerPublish:  10000,
                        publishingEnabled: true,
                        priority: 10
                    });

                    the_subscription.on("started",function(){
                        console.log("subscription started");
                        callback();
                        console.log("callback done");
                    }).on("keepalive",function(){
                        console.log("keepalive");

                    }).on("terminated",function(){
                        console.log(" TERMINATED ------------------------------>")
                    });

                }    
            ]);

I suppose something like : 

on("down",function(){
                        console.log("keepalive");
 }).

Might exists, but I can't find the correct syntaxe on the web.

Special thx to erossignon, I've read topics on the subject and you seems really invest and helpfull.

Loïc

PS : (sorry for my english)

avato92, 2019-03-05:
> Hi,
> I use node-opcua since a few weeks now. It's really powerfull and exactly what I was looking for.
> I still have some issues finding documentations on this product. Maybe you can tell me where I [...]

http://node-opcua.github.io/api_doc/0.2.0/classes/Subscription.html#events