node-opcua/node-opcua #117: 'MessageBuilder : ERROR DETECTED IN event handler'

zombinary, 2015-09-11:
I worked with V0.0.41 and after an update to V0.0.48 i get the same ERROR.

I call 'readVariableValue' with an 'selectNodeId' which not exist on the OpcUa Server. I get an 'err = 0'  and the console output
```
MessageBuilder : ERROR DETECTED IN event handler
[TypeError: Cannot read property 'value' of null]
```

Is there a Bug in the function 'readVariableValue' or i must resist the object {"name":"BadNodeIdUnknown"}.

my code:
```
node.session.readVariableValue(node.selectNodeId, function(err,dataValue,diagnostic){
      			if (err) {
     	     			console.log("ERROR\ ".red.bold + "readVariableValue:\ "+err);
          			console.log(dataValue);
	         		console.log(diagnostic);
	         		console.log("lalala");
//	         		callback(err);
         		}else{
                             console.log("\t value:\t\t ".yellow + dataValue[0].value.value.toString().green);
```

thanks for the help

erossignon, 2015-09-14:
you probably need to check for dataValue[0].statusCode .
```javascript
console.log(dataValue[0].toString());
if (dataValue[0].statusCode === opcua.StatusCodes.Good) {
    console.log("\t value:\t\t ".yellow + dataValue[0].value.value.toString().green);
}
```

zombinary, 2015-09-16:
thanks for your help

closed 2015-09-16 by zombinary