node-opcua/node-opcua #118: Status codes 'Bad' and 'Uncertain'

janschatz, 2015-09-15:
Shouldn't there be the status codes ```Bad``` (0x80000000) and ```Uncertain``` (0x40000000) without further specifications?

erossignon, 2015-09-15:
Could you elaborate ?

janschatz, 2015-09-15:
I am developing a gateway between OPC UA and a proprietary M2M protocol. I need to translate the qualities of the proprietary protocol to OPC UA status codes. I'd like to use more general codes like ```Bad``` and ```Uncertain```, because the mapping does not perfectly match - of course. As I understood these are allowed in OPC UA, aren't they?

```javascript
 exports.StatusCodes = {
  Good: { name:'Good', value: 0, description:'No Error' }
,                                Uncertain: { name:                              'Uncertain' , value: 0x40000000  ,description: "Uncertain"}
,                                      Bad: { name:                                    'Bad' , value: 0x80000000  ,description: "Bad"}
,                       BadUnexpectedError: { name:                     'BadUnexpectedError' , value: 0x80010000  ,description: "An unexpected error occurred."}

```

I just tried adding these codes and now any ```WriteRequest``` to my OPC UA server results in ```Bad```. \*confused\*

```
WriteResponse {
  responseHeader:
   ResponseHeader {
     timeStamp: Tue Sep 15 2015 16:48:50 GMT+0200 (Mitteleuropäische Sommerzeit),
     requestHandle: 1323,
     serviceResult: StatusCode { value: 0, description: 'No Error', name: 'Good' },
     serviceDiagnostics:
      DiagnosticInfo {
        namespaceUri: null,
        symbolicId: null,
        locale: null,
        localizedText: null,
        additionalInfo: null,
        innerStatusCode: null,
        innerDiagnosticInfo: null },
     stringTable: [],
     additionalHeader: null },
  results: [ StatusCode { value: 2147483648, description: 'Bad', name: 'Bad' } ],
  diagnosticInfos: [] } <-- It was good before!
```

erossignon, 2015-09-16:
OPCUA 1.0.3 part 4 page 147 defines Table 173 – Common Operation Level Result Codes with Bad and Uncertain.

janschatz, 2015-09-16:
Thanks!
It turned out that the ```Bad``` result was correct in my test. The ```Good``` result before had been wrong due to a bug in my call-back function.

closed 2015-09-16 by janschatz