node-opcua/node-opcua #121: connect endpoint without a certificate

zombinary, 2015-09-23:
I run the 'open62541' ./server_static without a certificate and get an error message, the session is not created.
```
MessageBuilder : ERROR DETECTED IN event handler
{ [AssertionError: serverCertificate instanceof Buffer]
  name: 'AssertionError',
  actual: undefined,
  expected: undefined,
  operator: undefined,
  message: 'serverCertificate instanceof Buffer',
  generatedMessage: false }
AssertionError: serverCertificate instanceof Buffer
    at OPCUAClient._activateSession (/usr/local/lib/node_modules/node-opcua/lib/client/opcua_client.js:377:5)
    at /usr/local/lib/node_modules/node-opcua/lib/client/opcua_client.js:540:18
    at /usr/local/lib/node_modules/node-opcua/lib/client/opcua_client.js:227:9
    at modified_callback (/usr/local/lib/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:730:24)
    at process_request_callback (/usr/local/lib/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:70:5)
    at _on_message_received (/usr/local/lib/node_modules/node-opcua/lib/client/client_secure_channel_layer.js:108:5)
    at emit (events.js:118:17)
    at MessageBuilder._decode_message_body (/usr/local/lib/node_modules/node-opcua/lib/misc/message_builder.js:443:22)
    at MessageBuilderBase._feed_messageChunk (/usr/local/lib/node_modules/node-opcua/lib/misc/message_builder_base.js:183:18)
    at null.<anonymous> (/usr/local/lib/node_modules/node-opcua/lib/misc/message_builder_base.js:46:14)
```

The client is create with following options
```
var options = {			
            		securityMode: "NONE",
            		securityPolicy: "None",
            		serverCertificate: null,
            		defaultSecureTokenLifetime: 2000
    			};
        			
client = new opcua.OPCUAClient(options); 
client.connect(endpoint,callback);
```

How can i connect the client without a certificate?

erossignon, 2015-09-23:
this problem relates to #67 and should be now fixed ( see 3375a6125413a805b264ca74fff3d34539248d92 ). Can you check with the latest master ?

zombinary, 2015-09-24:
ok check it works fine 
with version 'node-opcua v0.0.48' and 'open62541 v0.1.1'
thx

closed 2015-09-24 by zombinary