node-opcua/node-opcua #122: Connecting simple_server to UAGateway

indigodv, 2015-09-24:
Hi, I'm trying to connect the simple_server.js from the bin folder to UAGateway from unified automation. 
Seem like I cannot connect, 
 I always got this server status. 
[image]

I'd already added the certificate on the UAGateway Administration, but still get the same server status.
Any Idea on what I missed??

[image]

erossignon, 2015-09-24:
you could active some debug output on simple_server
simply set the DEBUG environment variable to "ALL"

dos:
```
set DEBUG=ALL
node bin\simple_server
```

bash:
```
export DEBUG=ALL
node bin/simple_server
```

indigodv, 2015-09-28:
still no luck.

erleeman, 2016-02-16:
No Problem to connect UAGateway or UAExpert.

[image]

Do you use the correct endpointUrl pointed out by the simple_server ?

closed 2017-01-09 by erossignon