node-opcua/node-opcua #123: Monitor multiple items

cremedalder, 2015-09-29:
Hi all.

I downloaded the dashboard example and I want to monitor multiple items.

If I simply add another monitored item like:

...........
var monitoredItem = the_subscription.monitor(
        {
            nodeId: "ns=4;b=1020ffaa",
            attributeId: 13
        },
        {
            samplingInterval: 100,
            discardOldest: true,
            queueSize: 100
        });
    monitoredItem.on("changed", function(dataValue){
         console.log(" Speed " +  JSON.stringify(dataValue));
        io.sockets.emit('message', {
            value: dataValue.value.value,
            timestamp: dataValue.serverTimestamp
        });
    });

////////////////////ITEM 2
var monitoredItem2 = the_subscription.monitor(
        {
            nodeId: "ns=4;b=1030ffaa",
            attributeId: 13
        },
        {
            samplingInterval: 100,
            discardOldest: true,
            queueSize: 100
        });
    monitoredItem2.on("changed", function(dataValue){
         console.log(" value has changed " +  JSON.stringify(dataValue));
        io.sockets.emit('message', {
            value: dataValue.value.value,
            timestamp: dataValue.serverTimestamp
        });
    });

........

then I only get the first console log approx. 2 times and then i  run into timeout:

Speed 1029081

 Timeout .... waiting for response for  { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=1;i=1061592575
 timeStamp                     /* UtcTime                          */: 2015-09-29T17:34:23.127Z
 requestHandle                 /* IntegerId                        */: 13               0xd
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */:
 timeoutHint                   /* UInt32                           */: 10000               0x2710
 additionalHeader              /* ExtensionObject                  */: null
};
 Timeout .... waiting for response for  { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=1;i=1061592575
 timeStamp                     /* UtcTime                          */: 2015-09-29T17:34:26.126Z
 requestHandle                 /* IntegerId                        */: 14               0xe
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */:
 timeoutHint                   /* UInt32                           */: 10000               0x2710
 additionalHeader              /* ExtensionObject                  */: null

Any ideas?

erossignon, 2016-01-18:
@cremedalder , sorry for a late reply on this. I have tried to highlight the issue you raised in a dedicated unit test. I cannot reproduce the issue you raised, May be I missed something or it has been fixed already. Can you tell us if you're still having the issue ?

erossignon, 2016-03-12:
@cremedalder  ?

cremedalder, 2016-03-14:
Hi Etienne.

I try to remember ....

I think after a long period of try and error we established a stable connection.

Here are the working and not working values:

WORKING:
requestedPublishingInterval: 1000,
            requestedLifetimeCount: 1000,
            requestedMaxKeepAliveCount: 1,
            maxNotificationsPerPublish: 10,
            publishingEnabled: true,
            priority: 10

samplingInterval: 500,
            discardOldest: true,
            queueSize: 200,

NOT WORKING:
requestedPublishingInterval: 1000,
            requestedLifetimeCount: 10,
            requestedMaxKeepAliveCount: 2,
            maxNotificationsPerPublish: 10,
            publishingEnabled: true,
            priority: 10

samplingInterval: 500,
            discardOldest: true,
            queueSize: 1,

And we had no solutions for monitoring multiple items with one subscription. Is there a way to do that?

What are the secrets behind these settings and why are there no problems by using products like UAexpert for example (https://www.unified-automation.com/de/produkte/entwicklerwerkzeuge/uaexpert.html). There I don`t have to adjust these pararmeters and it is simply working.

At the end the responsible people decided to go for a commercial product for the exhibition presentation ☹

Recently I tested node-red with node-opcua nodes. Looks promising.

Kind regards,

Hilmar.

Hilmar Hermens | +492562709430

Statutory/Registered Address: Jöbkesweg 3, 48599 Gronau, Germany
Company Registration: Amtsgericht Coesfeld HRB 5749, VATID-No.: DE 123764969
General Manager: Rolf Hemminga

erossignon, 2016-03-14:
> And  we had no solutions for monitoring multiple items with one subscription. Is there a way to do that? 
see #69

closed 2016-03-14 by erossignon