node-opcua/node-opcua #125: missing _auto_generated_ServerStatus.js

zombinary, 2015-10-09:
I switched from V0.0.48 to V0.0.49 (npm uninstall/install). 
First Problem is that the _auto_generated module is missing.
```
root@monkey:/usr/local/lib/node_modules/node-opcua/bin# node simple_server.js 
Warning : using memcpy : OK
FAST: using node-expat
module.js:338
    throw err;
    ^
Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
```
I copy a backup file '_auto_generated_ServerStatus', that was auto generated by V0.0.48 in the folder
'_/generated'.
After this, the 'simple_server.js' get a problem with the 'event.js' module

```
root@monkey:/usr/local/lib/node_modules/node-opcua/bin# node simple_server.js 
Warning : using memcpy : OK
FAST: using node-expat
  server PID          : 24522

registering server to :opc.tcp://monkey:1234/UADiscovery
 SOCKET ERROR  connect ECONNREFUSED CLIENT  Transport 0 undefined undefined undefined
error  connect ECONNREFUSED
     warning : cannot register server into registry server

events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: String cannot be coerced to a nodeId : 2150
    at coerceNodeId (/usr/local/lib/node_modules/node-opcua/lib/datamodel/nodeid.js:246:19)

```

With my own code i can connect to an opcua62541 server.
Who creates the '_auto_generated*.js ' files?
I couldn't find a script therefor?

erossignon, 2015-10-10:
I would not recommend to copy generated file from one version to an other as the generation engine may have changed between version. The best approach is to delete the ```_generated_``` and regenerate them. This can be achieved by running the the following script in the node-opcua folder:

```
node bin/generate_opcua_classes.js --verbose --clear
```

erossignon, 2015-11-05:
```
node bin/generate_opcua_classes.js --verbose --clear
```

zombinary, 2015-11-05:
Sry that i was long time offline.
I generated the files automaticle with your command. But event.js stopt this by an unhandled error event.

```
 generating  HistoryModifiedData  in  /usr/local/lib/node_modules/node-opcua/_generated_/_auto_generated_HistoryModifiedData.js
FAST: using node-expat
done
events.js:85
      throw er; // Unhandled 'error' event
            ^
Error: String cannot be coerced to a nodeId : 2150
    at coerceNodeId (/usr/local/lib/node_modules/node-opcua/lib/datamodel/nodeid.js:246:19)
```

currently i worked in another project ,  so i checked this later. Maybe it is a Problem with the event.js version.

closed 2015-11-05 by zombinary

erossignon, 2015-11-05:
Yes, this look wierd, => runing node-opcua on Travis does'nt exhibit the problem
You could also try to update the package on your local node-opcua repo clone 
```
npn install
```
Alternatively, reinstall a fresh copy of node-opcua

eddso, 2015-11-05:
Hi, same here on a fresh installation

<pre>
generating  HistoryModifiedData  in  /mnt/ENE-1_PRJ/_WORK_/IGW936/node/node-opcua/node_modules/node-opcua/_generated_/_auto_generated_HistoryModifiedData.js
FAST: using node-expat
done

events.js:72
        throw er; // Unhandled 'error' event
              ^
Error: String cannot be coerced to a nodeId : 2150
    at coerceNodeId (/mnt/ENE-1_PRJ/_WORK_/IGW936/node/node-opcua/node_modules/node-opcua/lib/datamodel/nodeid.js:246:19)
</pre>
node-opcua v0.0.49

erossignon, 2015-11-05:
Let's be more specified so we can reproduce the behavior you found;

```
$>mkdir mytest
$>cd mytest
$>node --version
0.12.2
$>npm install node-opcua
...
done
``` 

What version of node ?
What OS ?
Can you provide the step by step instruction to reproduce ?.

eddso, 2015-11-06:
OK today i installed version 0.0.49 again and it works. Steps like you posted. 
(yesterday i successed only with 'npm install node-opcua@0.0.48')
<pre>
node: 0.10.40
os: debian 6
locale: german
</pre>
thx. closed for me.

markovip, 2015-12-08:
Same issue.. 
node-opcua v0.0.50
node v4.2.3

rymoore, 2017-02-14:
installed the latest version on Mac and can't get past this issue:
```
Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
    at Function.Module._resolveFilename (module.js:337:15)
    at Function.Module._load (module.js:287:25)
    at Module.require (module.js:366:17)
    at require (module.js:385:17)
    at Object.<anonymous> (/Users/Ryan/SourceCode/opc-test/opcua-commander/node_modules/node-opcua/lib/server/server_engine.js:69:20)
```

erossignon, 2017-02-15:
I guess you're using old version of node ( 4.x ?) ? please update to node V6 or V7

rymoore, 2017-03-08:
I've just updated node on my Mac to v7.7.2 and now get a new error 

```
module.js:472
    throw err;
    ^

Error: Cannot find module 'schemas/39394884f696ff0bf66bacc9a8032cc074e0158e/ServerState_enum'
    at Function.Module._resolveFilename (module.js:470:15)
    at Function.Module._load (module.js:418:25)
    at Module.require (module.js:498:17)
    at require (internal/module.js:20:19)
    at Object.<anonymous> (/Users/Ryan/SourceCode/opctest2/node-opcua/lib/client/client_session_keepalive_manager.js:10:19)
    at Module._compile (module.js:571:32)
    at Object.Module._extensions..js (module.js:580:10)
    at Module.load (module.js:488:32)
    at tryModuleLoad (module.js:447:12)
    at Function.Module._load (module.js:439:3)

```