node-opcua/node-opcua #128: node_crawler "already visited"

nilsga, 2015-10-13:
I'm not sure if it is intended or not, but when crawling the address space, the crawling results in an array where elements can be of type string with the value " already visted....". Should these elements be removed before the crawler returns the array of nodes, or is it up to the client to handle these nodes? It kind of makes it awkward to process the array, since one have to take into consideration that the elements in the array is of different javascript types.

nilsga, 2015-10-13:
https://github.com/node-opcua/node-opcua/blob/012bafd2dad1e1a871045fbf5559cfc8328ce528/lib/client/node_crawler.js#L541

erossignon, 2015-10-18:
@nilsga , this line can be removed.

closed 2015-12-02 by erossignon