node-opcua/node-opcua #130: Questions: Auto-generated code

mpostol, 2015-10-17:
* Do you have auto-generated code ?
* Do you have special tool (application) to create this code or it is generated by the environment ?
* Where I can find source files for auto-generated code hopefully XML file for auto generated code ?
* Do you think it make sense to create a common place to manage this stuff ? Rel 1.03 is underway.

erossignon, 2015-10-20:
node-opcua uses auto-generated code:

**Request/Response data structures:**
- all the Request/Response data structures are automatically generated.
- The generation takes places during installation of the node-opcua module,
- for ease of use and concision, class schemas have been defined  in json rather than xml, see [here](https://github.com/node-opcua/node-opcua/blob/master/schemas/Argument_schema.js) or [here](https://github.com/node-opcua/node-opcua/blob/master/schemas/AttributeOperand_schema.js).As you can see JSON like structure are far more concise than their XML equivalent.
- the process of creating the json file from "bsd" file was intentionaly manual, as a learning curve to understand the underlying meaning of each data structure,  
- regeneration can be done by using this command:
```node bin/generate_opcua_classe --force```
- code regeneration uses a in-house templating scripts
- generated classes are stored in the ```_generated_``` folder

**Standard AddressSpace:**
- The AddressSpaces namespaces is generated by loading one or more ```NodeSet2.xml``` files at runtime. ( for example the [```simple_server.js```](https://github.com/node-opcua/node-opcua/blob/master/bin/simple_server.js#L55-L94)  loads the standard namespace and the DI namespace)
- custom behavior is then added to the UA Object at runtime using a binding mechanism. [see here to see how the server object is bound for instance](https://github.com/node-opcua/node-opcua/blob/master/lib/server/server_engine.js#L341)
- this process is rather different to the mechanism used by other opcua based on compiled languages such as C/C++/Java/.NET , and is made possible because javascript is a [dynamic programming language](https://en.wikipedia.org/wiki/Dynamic_programming_language) that allows you to create new object type and classes at runtime.
For this reason, there is a limited need to have external tools to generate boilerplate code prior compilation, although this may be useful.

mpostol, 2015-10-20:
I have reviewed what you said, but still cannot find list of build-in (e.g. UInt64) and standard data types (Structure) for example. I know they are in NodeSet but must be recovered. I am working on binary encoding/decoding for pub/sub stuff (Part 14) and there must be a long 'íf'. In this case dynamic programming language features seems to me not very useful. I am also looking for list of base and selected standard types for Variables and Objects.

Json you pointed out is an instance of program (provides behavioral description), but xml is only instance of document - we cannot compare both. Json may also be auto-generated as well.

erossignon, 2015-10-20:
built-in are normative and can be hardcoded 
definition is [here](https://github.com/node-opcua/node-opcua/blob/master/lib/misc/factories_builtin_types.js)

mpostol, 2015-10-20:
Yes it must be hardcoded, no doubts, but part of the code may be auto-generated. For example:
* [Here](https://github.com/mpostol/OPC-UA-OOI/blob/master/SemanticDataSolution/UANodeSetValidation/XML/BuildErrorsCollection.xml) you have XML file containing errors related to validation of NodeSet document;
* [Here](https://github.com/mpostol/OPC-UA-OOI/blob/master/SemanticDataSolution/UANodeSetValidation/BuildErrors1.cs) you can find the auto-generated code to be used in the C# program to report the errors.

XML can be used in any development environment, your Json and my C# usage is limited. There are many common definitions that can be useful across the languages. Having common definitions we could improve re usability not only for our projetcs.

closed 2016-07-18 by erossignon